f4exb/sdrangel #97: Putting in zeroes in delta-Frequency in demodulators gives inconsistent results

texasyojimbo, 2017-11-17:
Release: 3.8.2
Platform: Windows 10 
Source: LimeSDR 

Issue description: When I am using the AM demodulator and SSB demodulator, I will sometimes go to the delta-F box and start typing 0 (zero). Sometimes this sets the delta-F to 0,000,000.000 Hz. Other times it results in -9,551,616 Hz (and sometimes values in between depending on which significant digit is being edited). This may be present in others but hasn't been tested.

Reproduction steps: 

1. Start SDRAngel

2. Set LimeSDR channel 0 as source.

3. Start streaming (doesn't matter whether NCO is on or off).

4. Add demodulator as sink

5. Click below the main frequency in the waterfall, delta-F in the demodulator will change automatically to correct value (negative).

6. Go to the delta-F box in the demodulator, click on the largest significant digit (MHz, left-most value).

7. Start typing zeroes.

8. delta-F will change to (negative) -9,551,616 Hz.

9. Zero out delta-F manually using arrow keys.

10. Click above the main frequency, delta-F will change automatically to correct value.

11. Go to the delta-F box in the demodulator, click on the largest significant digit.

12. Start typing zeroes.

13. delta-F will change to +0,000,000 Hz (correct value).

Let me know if you'd like screenshots or WebEx where I can demonstrate.

texasyojimbo, 2017-11-17:
Changing decimation values does not seem to change value (always 9,551,661).
Screenshot attached

[image]

f4exb, 2017-11-17:
Maybe a bug. Using number keys should not do anything. It is meant to be used with mouse or arrow keys.

texasyojimbo, 2017-11-17:
Looking at the code (for ValueDialZ.cpp) I am going to go out on a limb here and suggest that there ought to be an absolute value used somewhere around line 580.

If I had more time to code and build, I would test this hypothesis.

f4exb, 2017-11-18:
Firstly `QChar('.')` is (hard)coded in many places. This should be changed to the locale's group separator. This is in both ValueDial and ValueDialZ

f4exb, 2017-11-20:
Implemented in v3.8.4

closed 2017-11-20 by f4exb