f4exb/sdrangel #98: Request: Some documentation on how to write plugins for demodulators

texasyojimbo, 2017-11-17:
Hi Eduoard,

Given the FOSS code out there for demodulating CW, PSK, and Bell 202 (Packet) I was wondering if it would be practical to write a custom plugin (possibly resulting in a pull request) to handle some common HF digital modes. So that one need not have to open MultiPSK or HRD or MixW to start listening in.

I see some documentation for the existing sinks as well as ReadmeDeveloper.md.

Do you have any other documentation that might be useful? Particularly if one wants to build a plug-in without re-building everything from source?

f4exb, 2017-11-18:
To be honest this project lacks documentation however the starting point when taking it over was absolute zero.

I looked at the ReadmeDeveloper.md and it is a bit outdated although the general picture is still valid. A large refactoring is taking place currently to extract the GUI parts from the rest so the details are not accurate anymore.

There is also an UML model in doc/model/SDRAngel.mdj. This is a [Star UML](http://staruml.io/) model but I haven't checked it recently so it could be even more outdated.

Usually when I want to write a new plugin I take an existing one as a base. The SSB demod could be a good starting point in the case you describe. The important thing to keep in mind now is that the GUI class can have read only access to the demod (it could also be read/write if it does not disturb the multi-threading) but not the opposite and that anything that goes from GUI to demod should go via the Settings class and message passing (thus always thread safe). If something needs to be passed back to the GUI this has to be via message passing after checking that the GUI message queue pointer is not zero (it will be zero in server mode).

f4exb, 2019-04-28:
dropped

closed 2019-04-28 by f4exb