f4exb/sdrangel #99: Gain is low when starting from a known-good preset

martywittrock, 2017-11-18:
I'm seeing an issue where I have a 'known-good' preset and everything seems to adjust right for frequency, sampling, gain settings, ect, but when I start the receiver the gain seems low. It's not until I adjust the sampling by 1 lower digit that the gain pops back up and noticeably higher, yet the same sampling rate.

brendanthebig, 2017-11-21:
I am having the same issue but I can get the gain level back to what I expect by adjusting the LP filter. Even putting it back to the preset level, the apparent gain is still ok. Considering that Marty is adjusting sample rate and I am adjusting LP filter to correct the gain, I think it might have something to do with the dreaded calibration procedure ????

brendanthebig, 2017-11-22:
Just a thought...Is calibration performed when a preset is loaded or only when a change is made to the sink or source settings ? Is calibration performed when a stream is started ?

f4exb, 2017-11-22:
Hello,

there are two calibrations: LMS_Calibrate and LMS_SetLPFBW. They are done in a single place but in many conditions. LMS_SetLPFBW as the name implies is done only if a change occurs in the LP filter. The main calibration LMS_Calibrate takes place if after all changes have been applied then a change in the clock generator takes place and also systematically in the following cases: 

  - change to auto gain
  - change to any of the gains in manual gain mode and if gain mode is switched to manual
  - host sample rate or hardware decimation/interpolation
  - change in antenna switch
  - change in center frequency
  - switch to external clock or back to internal clock 

In any case when you start streaming or load a preset all changes are activated hence calibration will take place.

martywittrock, 2017-11-22:
Edouard,

I was running my LimeSDR on the HF band (75m and 40m) and I noticed that the gain was very low - to the point that I tried an older version of SDRAngel (3.7.6-1) and from the same presets and what seemed magically my LimeSDR receive signal strength improved greatly over 3.8.3-1. So at the moment I've gone back to 3.7.6-1 since my receive is better. But there is definitely an issue with receive gain with the 3.8.3-1 release. I will try the newest release 3.8.4.7 and see if there's an improvement. but just reporting the gain issue since it is definitely there.

martywittrock, 2017-11-22:
Just tried 3.8.4.7 and it seems to be the same - has low gain in receive. I'm using a 5.0 MS/s and LPF at 100 MHz running H=32, S=32, FIR 2.500. There is definitely a difference in receive signal strength between 3.7.6.7 (3.7.6-1) and the prior release 3.8.3.7 and this latest release 3.8.4.7. The two latest releases are acting the same with the low receive signal strength against 3.7.6.7 that seems very much normal running the same receive preset that I've listed here.

f4exb, 2017-11-23:
There has been some code changes to do the calibration in the right conditions and this was done according to LimeSDR development team recommendations and I believe this is now correct. I have found no issues related to this code and in fact it is working much more smoothly.

Other than this and equally important are the changes done in the builds particularly the Windows build. Until 3.8.0 this was using LimeSuite 17.09.1 (3.7.8 was still using 17.09.1 this was a mistake) so when running 3.7.6 you are in fact using the LimeSuite 17.09.1 code and in revisions from 3.8.0 and above this is 17.10.0. The 17.10.0 needs the latest firmware (2.11) and you will have to upgrade your LimeSDR to be compatible.

In the next version (3.8.5) LimeSuite will have to be upgraded again to fully support LimeSDR mini (see #100).

brendanthebig, 2017-11-23:
I just upgraded my LimeSDR to latest firmware and fpga. Now I can't change freq in sampling device (main and nco) or change LP...

martywittrock, 2017-11-25:
Edouard,

I switched gears and purchased a UDOO Advanced board and installed Ubuntu 17.10 on it. I followed the directions, initially, to install SDRAngel from a .deb (for Ubuntu) installation but could not launch the app doing:

sudo /opt/sdrangel/bin/sdrangel

...Didn't start, but when I looked in that location it looked like a document (in FileManager) rather than an app. 

So then I compiled from source by taking the entire github repository and cloning it. I then did a build:

sudo mkdir build && cd build && cmake ../ && make

...and it built perfect - no errors. I then did:

sudo make install

...and it installed perfect. I then went to the /usr/local/bin/ subdirectory and found the same thing as before - a file that looks like a document called 'sdrangel' and not an app-like icon. I even tried launching it using:

sudo /usr/local/bin/sdrangel 

...and it didn't launch.

I've followed the directions to the letter and I've tried installing by two methods, .deb and a full compile, make, make install and I'm not able to make the SDRAngel application. From what I've written here, am I doing something really wrong here? If you can, let me know the steps to build SDRAngel or if I've missed a step maybe I can recover. Please let me know, because I want to use a Linux build on my UDOO Advanced board to drive the LimeSDR for two reasons: 1.) To check performance differences to my Win64 LattePanda, and 2.) Just to see if there are performance differences.

Let me know - I really want to play with this in Linux now and I think I've done everything right to make it happen. Again, let me know if I've omitted anything, Edouard - TNX es 73,

de Marty, KN0CK

martywittrock, 2017-11-25:
...One other note, I did make sure the dependencies were there - LimeSuite and SoapySDR...

73 de Marty, KN0CK

martywittrock, 2017-11-25:
Edouard,

I also tried a 'ldconfig' and then tried launching again and I was met with the following on the Terminal session:

"...QStandardPaths: XDG_RUNTIME_DIR not set, default to 'tmp runtime-root'
No protocol specified
QXcbConnection: Could not connect to display :0
Aborted..."

Hope this helps with the above- TNX es 73,

de Marty, KN0CK

martywittrock, 2017-11-27:
I've got a LimeSDR setup now for Linux running on a UDOO Advanced card loaded with Ubuntu 17.10 and compiled/ran the latest release sdrangel_3.8.4-1_amd64_artful.deb on that card and got it working properly once I had all the dependencies determined (LimeSuite, SoapySDR and others). Now that I have a good testing set, I will try an earlier release like I have for Windows to see if there's a difference or not using Linux. I am still of the opinion that there is a receive gain difference with older versions of SDRAngel as compared to the latest release, but I'll work that this evening and let you know how my older compile works...Stay tuned...

martywittrock, 2017-11-28:
As much as I hate to say this, there is a gain difference between the versions 3.7.6-1 and the current version 3.8.4.7. I can take my LimeSDR on the latest release and try to tune the 80/75m band (3.5 MHz to 3.99 MHz) at night and it's deaf. I can go back to 3.7.6-1 or Use SDRConsole V3.0 on my Windows machine and the LimeSDR looks great - I'm seeing signals and the receive activity is correct. the latest version makes my LimeSDR seem 'deaf' no matter that I'm setting the gain controls to maximum.

This is disappointing...SDRAngel was my 'go to' application for receive and transmit for the Lime. I can limp along with 3.7.6-1 forever, but the new features (like the headless design that's emerging) will not be in that version. Despite what the Lime Team has told you how to handle calibration and gain settings, there's something wrong with the latest release of SDRAngel. Even with the latest firmware installed using LimeUtil --update and seeing that the latest firmware is installed in LimeSuite makes no difference for the operation - the Lime still seems deaf. Yet I can take it across the room, plug it into SDRConsole V3.0 and it works great for receive.

What happened?

73 de Marty, KN0CK

f4exb, 2017-11-29:
This could be an issue with the analog lowpass filter since on 80m band (~ 3.5 MHz) you are very dependent on it being at ~26.5 MHz from the center of the analog passband if the LO is tuned at its lowest (30 MHz). I noticed that when loading a preset the LP value is not set correctly and remains as it was before. You have to fiddle with the LP filter thumbweels to make it effective. If the LP stays at a relatively low value then the 80m band is way out the filter limits. 

I also noticed that the value is the full passband. In the LMS7002M doc it says that the lowest value is 0.7 MHz but the lower limit given by LimeSuite is 1.4 MHz that is in fact +/- 0.7 MHz. It seems LimeSuite does the maths to translate complex lowpass to real lowpass around the center of reception.

martywittrock, 2017-12-03:
Edouard,

WOW!!! The 3.8.5 Windows release works WONDERFULLY AGAIN..!! Nice gain in the 80/75m band and well into the 40m band, too..! I haven't tried transmit but will tomorrow. I'm going to switch over to the UDOO card and see if I can compile it for Ubuntu 17.10 and check the performance - it's likely to be the same, but just wondering - More to follow on the Linux compile - stay tuned..! 

73 de Marty, KN0CK

f4exb, 2017-12-04:
Cool! so now this can be closed as implemented in v3.8.5

closed 2017-12-04 by f4exb