f4exb/sdrangel #101: SDRAngel 3.8.5 crashing when trying to start rx stream using LimeSDR[0:1] as source

brendanthebig, 2017-12-03:
Streaming runs ok when LimeSDR[0:0] is selected as source.

martywittrock, 2017-12-03:
I concur with Brendan's observation on this. When I tried to set up a second receiver, that part went fine. But when I wanted to start that second receiver everything froze (mouse, screen, and the receive activity that was happening on my first receive stream) and my LimeSDR was completely reset. So I unplugged my LimeSDR, waited about 30 seconds, plugged it back in set my presets and then started the first receive stream stream. Still had to fiddle with the major tuning (30.000) up to 50.000 and then back to 30.000 and then returned to the 75m band and my signal strength and tuning was good again. So this is an issue with the second instance of a LimeSDR stream.

73 de Marty, KN0CK

f4exb, 2017-12-04:
See disclaimer in the main readme: 

"LimeSuite library is difficult to implement due to the lack of documentation. The plugins should work normally when running as single instances. Support of many Rx and/or Tx instances running concurrently is considered experimental."

brendanthebig, 2017-12-04:
Edouard, I am only trying to run single rx instance. As the rf  input on LimeSDR[0:0] on my LimeSDR has been damaged, I have to use LimeSDR[0:1]. This has been working well in the previous few releases since you implemented direct selection of source/sink device channel on MIMO capable devices.

I hope this extra info is helpful.

Kindest regards,
Brendan Jenkins,
VK3WWB.

f4exb, 2017-12-10:
Problem is in Lime Suite:
```
Core was generated by `./sdrangel -p 8888'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f491a3614c8 in lime::ConnectionSTREAM::ReceivePacketsLoop(lime::ILimeSDRStreaming::Streamer*) () from /opt/install/LimeSuite/lib/libLimeSuite.so.17.10-1
[Current thread is 1 (Thread 0x7f4905eb2700 (LWP 25757))]
(gdb) bt
#0  0x00007f491a3614c8 in lime::ConnectionSTREAM::ReceivePacketsLoop(lime::ILimeSDRStreaming::Streamer*) () from /opt/install/LimeSuite/lib/libLimeSuite.so.17.10-1
#1  0x00007f491a33cd00 in std::thread::_Impl<std::_Bind_simple<std::function<void (lime::ILimeSDRStreaming::Streamer*)> (lime::ILimeSDRStreaming::Streamer*)> >::_M_run() () from /opt/install/LimeSuite/lib/libLimeSuite.so.17.10-1
#2  0x00007f494d154c80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007f494b9766ba in start_thread (arg=0x7f4905eb2700) at pthread_create.c:333
#4  0x00007f494cbc33dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

f4exb, 2017-12-10:
Same issue with LimeSuite 17.12.1:
```
Core was generated by `./sdrangel -p 8888'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007fa10275f4b8 in lime::ConnectionSTREAM::ReceivePacketsLoop(lime::ILimeSDRStreaming::Streamer*) () from /opt/install/LimeSuite/lib/libLimeSuite.so.17.12-1
[Current thread is 1 (Thread 0x7fa0edb80700 (LWP 27884))]
(gdb) bt
#0  0x00007fa10275f4b8 in lime::ConnectionSTREAM::ReceivePacketsLoop(lime::ILimeSDRStreaming::Streamer*) () from /opt/install/LimeSuite/lib/libLimeSuite.so.17.12-1
#1  0x00007fa10273ad00 in std::thread::_Impl<std::_Bind_simple<std::function<void (lime::ILimeSDRStreaming::Streamer*)> (lime::ILimeSDRStreaming::Streamer*)> >::_M_run() () from /opt/install/LimeSuite/lib/libLimeSuite.so.17.12-1
#2  0x00007fa13573dc80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007fa133f5f6ba in start_thread (arg=0x7fa0edb80700) at pthread_create.c:333
#4  0x00007fa1351ac3dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

f4exb, 2017-12-10:
It appears that due to the oddities of LimeSuite you cannot start the second stream alone. Starting just stream 0 will work and starting stream 0 then stream 1 will work also but neither starting stream 1 alone nor starting stream 1 before stream 0 will work. I tried on Rx side only but I suppose it is the same on Tx side.

brendanthebig, 2017-12-10:
Seems a bit strange as it works fine in version 3.8.4. Do you mind if I copy the contents of this thread to the Myriad Discourse forum for the LMS guys to have a look at ?

f4exb, 2017-12-10:
Hello Brendan, 

this is not so strange if you consider that between 3.8.4 and 3.8.5 the LimeSuite version used in the builds was upgraded from 17.10.0 to commit 0167e64. This commit: 742c73b - Streaming changes could be the cause judging by the commit message. Anyway it seems the streams are not independent which is not a problem if you use Soapy because with Soapy sreams are not independent either.

I am afraid this is a limitation rather than an actual bug. You may try to report this issue to Myriad Discourse if you think this could help.

Best regards, Edouard.

brendanthebig, 2017-12-10:
Thanks Edouard, considering my situation with my LimeSDR, I do intend to follow this up. I am becoming quite fond of SDRAngel and would hate for something trivial like this to stop my enjoyment. Hopefully the LMS team can shed some light on the problem. Maybe even come up with some documentation that is more helpful in resolving such issues.

Kindest regards,
Brendan Jenkins,
VK3WWB.

brendanthebig, 2017-12-10:
Posted to Myriad RF Discourse forum :- 

brendanthebig1h
@andrewback

Hi Andrew,

the current version of SDRAngel (3.8.5) is having some problems with the latest Limesuite release. If you get a chance, can you please have a look at the comments in the attached link to a thread on Edouards SDRAngel github issues list…

github.com/f4exb/sdrangel1
 
Issue: SDRAngel 3.8.5 crashing when trying to start rx stream using LimeSDR[0:1] as source

opened by brendanthebig on 2017-12-03
Streaming runs ok when LimeSDR[0:0] is selected as source. Currently using 3.8.5 Windows version on Win7 x64 system.
wontfix
Could you please direct us to the appropriate person to follow this issue up with.

Kindest regards,
Brendan Jenkins,
VK3WWB.

9600, 2017-12-10:
@brendanthebig you've just ATed someone else there... Will follow up on discourse.myriadrf.org.

brendanthebig, 2017-12-10:
Sorry and Thanks Andrew.

brendanthebig, 2017-12-11:
Hi Edouard, it looks like ignasj has fixed the issue with LimeSuite. 

https://github.com/myriadrf/LimeSuite/commit/04b57e0a247debb451b067b3fc486307627ba5b7

Hopefully that gets things working again.

Kindest regards,
Brendan Jenkins,
VK3WWB.

f4exb, 2017-12-12:
I tried the latest commit on LomeSuite master and  it appears to have fixed the problem. That's very  good news indeed !

brendanthebig, 2017-12-13:
Excellent...I'm looking forward to trying it out...

brendanthebig, 2017-12-16:
Hi Edouard, all working well now. Issue is resolved. A very big thank you to you.

closed 2017-12-16 by brendanthebig