f4exb/sdrangel #102: SDRAngel will not start on some windows 10 machines

Radiophilo, 2017-12-03:
Will not start at all and no error message is shown.
Windows logs an application error event "Exception code: 0xc0000005".

One example is a Lenovo X220 Thinkpad with Windows 10 x64.
Graphics chip is Intel HD 3000 and OpenGL Version is 3.1.

Please confirm OpenGL 3.X is supported and OpenGL 4 is not required.

f4exb, 2017-12-04:
I will guarantee only what I see in front of my eyes so on Windows the minimum OpenGL version required is 4.3

texasyojimbo, 2017-12-15:
Hi Eduoard:

I thought I would share my own experience as it may bear on this issue.

I find that sdrangel 3.8.5 does not work when I have this version of my graphics card driver installed:

Intel HD Graphics 400
12/12/2012
9.17.10.2932

According to GLView OpenGL Extensions Viewer, the OpenGL version is 4.0. Note that it is 100 percent compatible with OpenGL 4.0 and lower, but zero compatible with higher.

However, when I switch to the newer build of the driver (12/21/2015, 10.18.10.4358), sdrangel will run. This driver shows up in GLView OpenGL Extensions Viewer as being 100 percent compatible with 4.0, and partially compatible with 4.1, 4.2, 4.3 and 4.4 (it scores as 80% + compatible with 4.2 and lower for 4.3 and higher).

Unfortunately it seems like "it depends" is probably the right answer, but sdrangel definitely seems to be dependent upon some functionality added to OpenGL after 4.0. It definitely would not seem to require 4.4 or 4.5 however.

If anyone with a modern (post-2010 or so) machine has trouble starting sdrangel on Windows 10, the first thing they probably ought to do, is to try to upgrade their display driver (and barring that, the video card).

f4exb, 2017-12-15:
There seems to be a noticeable difference between Linux and Windows. On Linux version 3.0 would do. With Windows I only have a working example with 4.3 so I couldn't check for lower versions. The OpenGL version should appear in the log in debug mode when the OpenGL context is created in `GLSpectrum::initializeGL()`:

```c
	if (QOpenGLContext::currentContext()->isValid()) {
		qDebug() << "GLSpectrum::initializeGL: context:"
			<< " major: " << (QOpenGLContext::currentContext()->format()).majorVersion()
			<< " minor: " << (QOpenGLContext::currentContext()->format()).minorVersion()
			<< " ES: " << (QOpenGLContext::currentContext()->isOpenGLES() ? "yes" : "no");
	}
```
I get 3.0 in Linux and 4.3 in Windows for a working case.

f4exb, 2017-12-17:
Documentation updated in v3.8.6

closed 2017-12-17 by f4exb

f4exb, 2017-12-31:
It seems the Qt used to build the Windows 64 bit version does not have ANGLE support which is required to work on Windows systems with only the default OpenGL 1.1 Including a 32 bit build even if it misses some plugins might help.