f4exb/sdrangel #103: Missing drop down in sources window.

vsboost, 2017-12-05:
Hi there,
Seem to be missing the drop down window for source devices.
[image]

Windows 10 Surface pro.

Daniel-dk, 2017-12-06:
I'm getting the same on a win10 pro desktop

vsboost, 2017-12-10:
Think this is a high-dpi issue.

f4exb, 2017-12-10:
What do you mean by "high-dpi"? Even with no devices connected you should get "FileSource" and "SDRdaemonSource" in the list so I was thinking that in your particular installation these two plugins do not load correctly for some unknown reason.

f4exb, 2017-12-10:
If you can display it you may try to activate the log to file in debug mode with preferences -> logging menu. Then close and restart the application. On the next run it will log messages to file which may give indication if the plugins load correctly or not:
[image]

ChiliX, 2017-12-11:
Had exactly the same problem on a Retina MacBook Pro & Win10 in BootCamp. By switching screen resolution to 1920x1200 I could see the drop down list and select the source device.

antoine-roux, 2017-12-12:
Hello, I confirm the high-DPI issue: on a given machine (ArchLinux) the dropdown is displayed correctly with a 1.0 scaling but not on a 2.0 scaled display.

Salut à toi, sur les affichages haute densité (les Surface de Microsoft ont des écrans à 267 dpi) la dropdown n'est pas affichée. Le programme se comporte différemment en fonction de la densité de pixels. Le problème est tout récent, j'ai pu l'utiliser correctement avant que je le mette à jour il y a environ une semaine.
Sur Windows, essaye de jouer sur la taille de police système (Display -> Font size). Mets la à 200% pour voir apparaître le problème.
Sur Linux essaye de changer org.gnome.desktop.interface scaling-factor ou org.gnome.desktop.interface text-scaling-factor à 2.0

Bien à toi,
Antoine

vsboost, 2017-12-12:
[image]

Seems to fix it for me.

antoine-roux, 2017-12-12:
Yes, but it's still an issue on Linux.

I have two different behaviors on my machine:
When my primary screen is not an HDPI one, I run sdrangel with `QT_AUTO_SCREEN_SCALE_FACTOR=0 QT_SCREEN_SCALE_FACTORS="1;1;2" sdrangel`, everything runs smoothly (3 screens: first one and second one in standard DPI and the third in HDPI). And I can move the program between my screens, the DPI changes on-the-fly.

But curiously, with only one screen and using the native HDPI scaling, I can not get the dropdown working. Even I play with `QT_*` variables...
[image]

Cheers,
Antoine

f4exb, 2017-12-14:
It looks like the QT variables for HDPI support were introduced in Qt 5.6 and the Xenial (Ubintu 16.04) version is compiled with Qt 5.5 so I guess you will need at least Zesty (Ubuntu 17.04) that compiles with Qt 5.7 to have it work.

f4exb, 2017-12-14:
The source of the dialog is at sdrgui/gui/samplingdevicedialog.ui If anyone notices something missing for HDPI support they're welcome to provide a fix.

antoine-roux, 2017-12-16:
@f4exb I'm using Qt 5.10 on ArchLinux

f4exb, 2018-01-18:
This question would be better answered at: https://groups.io/g/sdrangel

closed 2018-01-18 by f4exb

f4exb, 2018-10-24:
Sorry nothing I can do it seems to be Qt business.

simonvik, 2018-10-24:
https://doc.qt.io/qt-5.6/qtlabscontrols-highdpi.html does seem to help on it, the UI renders somewhat correctly when setting that env, all icons are scaled and resolution is somewhat low but the dialog box works. Will try to play a bit more with this.

simonvik, 2018-10-24:
Its good enough for me to set QT_SCALE_FACTOR or QT_AUTO_SCREEN_SCALE_FACTOR and i think that ubuntu might do this too globaly somewhere.

f4exb, 2018-10-26:
I'll be adding this in the [main app code](https://github.com/f4exb/sdrangel/blob/master/app/main.cpp) for next version. It may be useful and anyway it does not seem to hurt:
```
    QApplication::setAttribute(Qt::AA_EnableHighDpiScaling); // DPI support
    QCoreApplication::setAttribute(Qt::AA_UseHighDpiPixmaps); //HiDPI pixmaps
```

ktoksoz, 2018-11-23:
I compiled master branch right now and can say that bug does not fixed.. It stil looks funny and dropdown does not have a down arrow to select.

f4exb, 2018-11-23:
Sorry, no issues here.

vsboost, 2020-06-20:
Thanks for fixing this. v4.14.9

[image]
[image]