f4exb/sdrangel #104: Unable to build on Zesty

RyanThaDude, 2017-12-05:
I've been banging my head against the wall trying to figure out why `sdrangel` won't build. It will error during make at this error

```
[ 40%] Building CXX object devices/limesdr/CMakeFiles/limesdrdevice.dir/devicelimesdr.cpp.o
/home/ryan/git/sdrangel/devices/limesdr/devicelimesdr.cpp: In static member function ‘static bool DeviceLimeSDR::setClockSource(lms_device_t*, bool, uint32_t)’:
/home/ryan/git/sdrangel/devices/limesdr/devicelimesdr.cpp:352:38: error: ‘LMS_CLOCK_EXTREF’ was not declared in this scope
         if (LMS_SetClockFreq(device, LMS_CLOCK_EXTREF, (float) extClockFrequency) < 0)
                                      ^~~~~~~~~~~~~~~~
devices/limesdr/CMakeFiles/limesdrdevice.dir/build.make:62: recipe for target 'devices/limesdr/CMakeFiles/limesdrdevice.dir/devicelimesdr.cpp.o' failed
make[2]: *** [devices/limesdr/CMakeFiles/limesdrdevice.dir/devicelimesdr.cpp.o] Error 1
CMakeFiles/Makefile2:850: recipe for target 'devices/limesdr/CMakeFiles/limesdrdevice.dir/all' failed
make[1]: *** [devices/limesdr/CMakeFiles/limesdrdevice.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```
I have the LimeSDR and RTL-SDRs and building with `dsdcc`. CMake command is
```
cmake \
-DLIBDSDCC_INCLUDE_DIR=/usr/local/include/dsdcc \
-DLIBDSDCC_LIBRARIES=/usr/local/lib/libdsdcc.so \
-DLIMESUITE_INCLUDE_DIR=/usr/include/lime/ \
-DLIMESUITE_LIBRARY=/usr/lib/x86_64-linux-gnu/libLimeSuite.so \
-DLIBRTLSDR_LIBRARIES=/usr/lib/x86_64-linux-gnu/librtlsdr.so \
-DLIBRTLSDR_INCLUDE_DIR=/usr/include \
../
```
While I can install it via the dpkg (it does work) I prefer to compile it.

f4exb, 2017-12-06:
You need the latest version of LimeSuite. LMS_CLOCK_EXTREF appeared in 17.09.0.

RyanThaDude, 2017-12-06:
That did it. At first I thought I had the latest version of LimeSuite but didn't. Thanks!

closed 2017-12-06 by RyanThaDude