f4exb/sdrangel #105: Feature request - Support SDRPlay RSP1A

unixpunk, 2017-12-08:
Hi, I can't seem to get my RSP1A to work in sdrangel, doesn't show up, but it does show up in another sdr software.  I assume something is different with RSP1 vs RSP1A and that's why.  Happy to help if possible.

f4exb, 2017-12-09:
It is probably an enumeration issue. I don't know how the RSP1A shows up in the list of USB devices as I do not own one. Apart from this one should also consider that the RSP1A has a 14 bit ADC vs a 12 bit ADC for the RSP1 so to take the full benefit the decimators should work on 14 bit input which in the end means it has to be considered as a different device not just a variant.

f4exb, 2017-12-09:
These are the vendor id, device ids recognized by libmirisdr-4. If anything different it will not work:
```c
static mirisdr_device_t mirisdr_devices[] = {
    { 0x1df7, 0x2500, "Mirics MSi2500 default (e.g. VTX3D card)", "Mirics", "MSi2500"},
    { 0x2040, 0xd300, "Hauppauge WinTV 133559 LF", "Hauppauge", "WinTV 133559 LF"},
    { 0x07ca, 0x8591, "AverMedia A859 Pure DVBT", "AverTV", "A859 Pure DVBT"},
    { 0x04bb, 0x0537, "IO-DATA GV-TV100 stick", "IO-DATA", "GV-TV100"},
    { 0x0511, 0x0037, "Logitec LDT-1S310U/J", "Logitec", "LDT-1S310U/J"}
};
```

unixpunk, 2017-12-09:
Ah OK, makes perfect sense.  lsusb for the rsp1a shows this: 1df7:3000.  SoapySDRUtil seems to core dump on me lately, so can't get you the additional details on the device, if that would help, at this point.  Working on it now.

unixpunk, 2017-12-09:
Here you go, let me know if I can help further.  Turns out it was the Pluto lib causing Soapy's issues.  Also, not sure where libmirisdr-2 gets used for other software, but it looks like Soapy interfaces directly with the SDRPlay API itself in case that helps or not, libmirsdrapi-rsp.so.
----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=1712xxxxxx  (omitted hex serial number)
  mir_sdr_api_version=2.110000
  mir_sdr_hw_version=255

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8, 9)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]
     * DabNotch Enable - DAB Notch Filter Control
       [key=dabnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 48] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 9] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

f4exb, 2017-12-13:
I'd be more comfortable if I had a unit to test. It should be on its way but I expect some delay due to re-stock in the shop. Seeing from a high level point of view it should be feasible to recognize the RSP1A specifically thanks to its specific USB signature and derive a specific flavour out of it so that maybe the same sdrplay plugin could be used if the features are not too different.

unixpunk, 2017-12-13:
Understand completely, I'm happy to help where I can, I'm not much of a coder though so I do appreciate all your effort on this project, its my new favorite SDR application!  Also, not sure where those default values above come from, but I wouldn't think one _wouldn't_ want the biasT option to be _enabled_ by default.  I think I did have some issues with that at some point where stopping/starting the device would flip on the biasT in a situation that it shouldn't be.

I have hackrf (w/portapack), various RTL's and a PlutoSDR if you ever need anything else tested.  Also have a KiwiSDR on the way.

f4exb, 2017-12-14:
Don't know about KiwiSDR I will have a look. I see a very good reason why you wouldn't like the biasT enabled by default: if your antenna port is connected to a folded dipole or a loop then you will short circuit the bias supply.

unixpunk, 2017-12-14:
KiwiSDR is more of a self-contained unit with a web interface like OpenWebRx.  Shows you 0-30mhz in the waterfall at once and allows 4 independent listeners.  I was thinking it would be cool if it had some ability on UDP, then maybe you could use sdrangel as the front-end software instead of the OpenWebRx it comes with.  I'll let you know what I find later this month, the software is on an SD card and openwebrx just launches whatever binary to control the radio itself, maybe there's some flags there that can be modified to allow access to I/Q data, that's my hope at least.  :)  I mainly got it because it will pair nice with my Wellbrook loop antenna, though the RSP1A does a surprisingly good job on HF too with the right gain settings.

And yes, exactly my point about biasT, should be off by default.  :)  The 2 notch filters could go either way, I have to enable them where I'm at because of a nearby transmitter on FM band even though I already have a 30mhz low-pass filter on the antenna.  Maybe because its active, picks up more.

f4exb, 2017-12-19:
Got the RSP1A now. I modified the limirics-4 library and implemented new udev rules and I can get something out but I think the gain controls should not be quite right, We'll see...

unixpunk, 2017-12-19:
Nice, thanks!  The RF gain should be set to 0 which is max, don't know why its opposite most gains.  If you set that to 0 and use AGC, you should be able to pick up something.  Make sure the one other option is set to ZERO-IF (I think it was) as well.  RF notch is AM/FM and DAB is, well, DAB.  :)  Sounds like I just need to pull/recompile libmirics-4 and recompile sdrangel against it then?

f4exb, 2017-12-19:
It seems not possible to support RSP1A at this point since controls specific to RSP1A are unknown. This ticket has been opened in libmirisdr-4 project to ask for help: https://github.com/f4exb/libmirisdr-4/issues/2

unixpunk, 2017-12-19:
Hm, the SoapySDRUtil output above doesn't help?  It has all the settings/values available.  I'm assuming libmirics-4 still relies on the SDRPlay API accessed via SoapySDRPlay plugin?  I'm following other issue as well.  Thanks again for your work!

f4exb, 2017-12-20:
No libmirisdr-4 uses direct connection via libusb. It does not use llibmirsdrapi-rsp.so which is binary only either. Soapy is not used at all in SDRangel.

unixpunk, 2017-12-20:
Ah OK, good to know.  I'll try to find some more information or help otherwise.  Thanks again!

idl0r, 2018-08-29:
It could make more sense to add SoapySDR support and benefit from it's device support. You wouldn't have to implement all those different kinds of devices it seems. I'd be interested in this as well.

f4exb, 2018-08-29:
SoapySDR cannot be supported just like another device. It is a set of devices so the concept of virtual device has to be introduced. This is not a priority at the moment.

f4exb, 2018-09-28:
Closed in favor of #221

closed 2018-09-28 by f4exb