f4exb/sdrangel #106: About drm signal in new sdrangel software

longyucheng123, 2017-12-13:
I hope to add the modulation and demodulation of the Drm signal in the new version. Thank you!

f4exb, 2017-12-13:
This is a great idea to have a DRM plugin. I hope it is not too difficult to implement a new plugin. I think it could be done along the lines of the analog voice demodulators (AM, FM, SSB or WFM) that could be used as a base. Do not hesitate to comment if you have any questions.

mpbraendli, 2018-02-18:
Hi,
https://github.com/Opendigitalradio/qt-drmplus could maybe help.
73 HB9EGM

f4exb, 2018-02-18:
Looks interesting thanks! Other repositories of the same author show work about DAB also.

mpbraendli, 2018-02-19:
Otherwise there's also https://github.com/kit-cel/gr-drm

f4exb, 2022-02-10:
dropped

AsciiWolf, 2022-07-22:
Any chance this could be implemented? It would be really great to have [DRM](https://www.drm.org/what-is-drm/) receiver in SDRangel. It could be based on [Dream](https://sourceforge.net/projects/drm/) software receiver.

srcejon, 2026-10-27:
Implemented in v8.

closed 2026-10-27 by srcejon