f4exb/sdrangel #107: Windows 32 bit?

dsound-zz, 2017-12-17:
I'm not clear how to install this on a Windows 32 bit (Win 10) machine.

dcrand2012, 2017-12-17:
I'm at a lost also. Tried install bats but no joy_

f4exb, 2017-12-17:
To build from source for Win32 you may try to follow the guidelines here: https://github.com/f4exb/sdrangel/blob/master/ReadmeWindowsBuild.md

Beware that is is not for the faint hearted. However the 32 bit version is normally the default when you install Qt Creator so you shouldn't have to fiddle with MSys2. A critical part is to get the dependencies right. It could help to use the same paths than in the .pro files.

The batch files are there just to copy the binaries after the build to an installation directory.

DL4APT, 2017-12-19:
I have an older Version (3.6.0) that is still 32 bit still laying around if that my help

f4exb, 2017-12-31:
Done in v3.9.0

closed 2017-12-31 by f4exb

yo8sxt, 2018-12-26:
hello,i wish to install sdrangel on a 32bit machine but i can,t find the v390 or v360 on the internet.please can you help me with a copy or direct me to a place were i can find a 32bit sdrangel?!