f4exb/sdrangel #108: Decode TETRA

jonas18z, 2017-12-18:
Is it possible to decode unencrypted tetra?
http://www.rtl-sdr.com/tetra-decoding-on-windows-with-telive/
http://tetra.osmocom.org/trac/

f4exb, 2017-12-19:
There is presently not such plugin although it would be nice (contributors welcome). However you can use SDRangel as a radio front end to the decoding part in GNUradio by piping I/Q samples into GNUradio via UDP. 

  - On SDRangel side use the "UDP Channel Source" channel plugin in I/Q mode (source for external programs)
  - On GNUradio use a UDP source block (source in a graph)

DG9BFC, 2017-12-19:
what about adding a "normal" support via virtual audio cable?!? then any use can setup any decoding software with sdrangel (like done in nearly all other sdrsoftware)

f4exb, 2017-12-19:
Virtual audio cable is usually not a very smart way to connect applications. Wisely enough GNUradio uses UDP to transfer samples and this could be used with SDRangel.

Anyway and on Linux only portaudio is used which should make some audio plumbing possible.

E3V3A, 2018-01-10:
@jonas18z You also found [this](https://github.com/sq5bpf/telive), right?

jonas18z, 2018-04-03:
Yes but I wanted it in de sdrangle

opabert, 2019-03-27:
> There is presently not such plugin although it would be nice (contributors welcome). However you can use SDRangel as a radio front end to the decoding part in GNUradio by piping I/Q samples into [...]

Try contact

opabert, 2019-03-29:
Like test new tetra plugin any contact with author possible?

gaspar1987, 2019-07-05:
hello  sur le  4.11.5  la crypto voice DMR TERMINALE sur dsd demodeur .

f4exb, 2020-08-07:
Someone pointed out this to me that can be interesting: https://gitlab.com/larryth/tetra-kit
It was also metioned in this RTl-SDR.com post: https://www.rtl-sdr.com/tetra-kit-a-new-open-source-tetra-decoder/

f4exb, 2022-02-10:
dropped

closed 2022-02-10 by f4exb