f4exb/sdrangel #109: Sdrangle6v don't work on W10x64

PD1LOI, 2017-12-19:
I have downloaded the last version 3.8.6 sdrangel64_v but this won't boot in 1709 version W10x64 build 16,299,125. How can this be?

DL4APT, 2017-12-19:
What processor/graphic card is in your PC? The newer Versions of SDRAngel needs at least OpenGL 4+ to run. My older Laptop has a HD3000 that officially only supports OpenGL up to 3.3 so I  can only run the Old 32 bit versions (3.6.0) that don't use newer features.

PD1LOI, 2017-12-19:
I have a Compaq 6710b 4 GB ram and 500 GB HDD

DL4APT, 2017-12-19:
If I'm not mistaken your Laptop has a GMA X3100 GPU inside the chipset. That GPU only supports OpenGL 2.1 and is now over 10 Years old. 
Maybe an older 32bit version might work. As working with lager amounts of SDR Bandwidth (for example to use the 60MHz of an Lime SDR) needs a quite fast system. So to get more performance out of an newer system i think the decision was made to use newer hardware acceleration features. This is at the cost of real old systems that would have quite poor performance  anyway. 
TLDR: to get newer systems performance from normal to good new features are used that get the performance on old systems from terrible to none.

f4exb, 2017-12-19:
I can only second this latest statement. If for any good reason a 32 bit version is needed I could look into producing one again but the target is clearly modern 64 bit systems.

PD1LOI, 2017-12-20:
Mobile Intel(R) 965 Express Chipset Family (Microsoft Corporation - WDDM 1.1)
Works fine on my Laptop.

f4exb, 2017-12-20:
OpenGL 1.1 is no way because it does not recognize programmable function shaders.
And... come on please! OpenGL 1.1 is 20 years old!
https://www.khronos.org/opengl/wiki/History_of_OpenGL

PD1LOI, 2017-12-20:
May be, but my laptop works better and faster than many low budget laptop
this time that many are sold at media companies. But right then but not.
SDRsharp also works fine with no problems including Dsd +

closed 2017-12-20 by f4exb

f4exb, 2017-12-30:
It might well be related to this: https://github.com/Alexpux/MINGW-packages/issues/3002

f4exb, 2018-01-01:
starting with v3.9.0 Windows 32 bit version is distributed again.