f4exb/sdrangel #110: Poor audio on braodcast FM (Win-10/64).

gm8arv, 2017-12-20:
I've been trying 3.8.6 for receiving broadcast FM and digital voice modes DMR/D-Star/C4/FM.  Decoding is very marginal on DMR, and only a little better on D-Star.  Frequency and levels appear correct as far as I can tell, but I can't find a guide to tell me what they should be.  Where can I ask questions and get help on this very promising software?  Yahoo or groups.io?  (I don't do Facebook).

As the audio was rather distorted when I could get it to work, I also checked broadcast FM and that sounded somewhat distorted too.  Should it be "near perfect"?  There's plenty of CPU and memory in hand.  Airspy Mini at 3 or 6 Msps.

Thanks!

f4exb, 2017-12-20:
There is documentation on the DSD demodulator for digital modes here: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demoddsd/readme.md you can find the link in the Wiki as well.

There is no such documentation for the Broadcast FM demodulator. The important point is that the RF bandwidth (RF BW) should not be lower than 160 kHz else it can be heavily distorted. You have to check the volume control also not to saturate the sound card.

gm8arv, 2017-12-21:
Thanks!  I didn't know about that resource, which needs to be more widely advertised.  Here's a sample screen-shot showing DMR from my hotspot, but direct from the local repeater it's similar.  I very rarely get a lock, and perhaps the large white ellipse indicates too?

Both are Airspy receivers, but I could take the IF from an IC-R8600 - it can output the IF as an audio source.  However, I can't seem to find an audio input as a sampling device, despite having set it in the Preferences, Audio, Audio Input.

[image]

f4exb, 2017-12-21:
Normally people who use Github know how to find the Wiki of a project and the readme files of the different plugins are linked from there.

So the remaining question is more about DSD than Broadcast FM in fact. Firstly you should keep the same scale for X (A1) and Y (A2) units on the scope and best is to choose "1" therefore the scale is +/-500 mUnits. So the ellipse is in fact a circle and I suppose it would look pretty good. Once you have adjusted the RF bandwidth so that all of your signal is in the bandwidth you can adjust the delta frequency so that the figure is centered and the FM deviation and/or the gain so that it approximately fills the square. Then your settings should be optimal for good decoding. Judging by the display it seems that you are synchronized (+DMRv) and you do receive voice (VOX) frames.

gm8arv, 2017-12-21:
Thanks!  

Yes, I hadn't noticed that the scope scales weren't equal as I had been swapping between different sources to try and confirm where the digital audio issue was.  Now having the proper monitoring I have made some minor adjustments, and D-Star decoding is sometimes readable.  Yes, it's not that good.  Many times the lock indicator stays grey rather than going green.  On DMR the decoding is almost non-existent, and while you can see something like the pattern of dots shown in the Wiki, it's not as sharp as clear as that, with perhaps less "dwell" time on the constellation points.  The lack of lock and the lack of decoding seem to be linked, as you would expect.

The one other common factor is the Airspy, although I've tried both a full-size and a mini.  I see that you can specify an audio input, but not as a source.  I could perhaps feed the 12 kHz IF from my IC-R8600 into SDRangel, as I know the IC-R8600 decodes D-Star perfectly.  Alternatively, I see there is a file input option.  Could anyone supply a file with DMR which is known to decode correctly in case that helps with further problem resolution?

I do now see the Wiki link at the top of the page.  It may still be worth setting up support on e.g. groups.io where users can help each other as I see the software could become very popular.  It leaves you free to concentrate on improving the software!

f4exb, 2017-12-21:
It took time for me to notice because I was focused on the XY diagram but there is certainly a problem with a baseband sample rate lower than 48 kHz. Here I see 46.875 kHz ("46.875 k" in the top left corner close to the frequency dial). This will certainly not work as the decoder expects 48 kHz samples. Therefore the baseband sample rate has to be larger (it is decimated and not interpolated). So I suggest that you use a software decimation of 32 instead of 64 increasing the baseband sample rate to 93.75 kHz.

There is no audio input as "sampling device" what you could possibly do but that would be on Linux only is to use named pipes. The audio input that you can see in the audio preferences is for transmitting. I though of creating plugins for audio in/out but always had other things to do with higher priority. This could be made a feature issue.

I had a quick look at groups.io. That's interesting however I could end up spending the same amount of time replying on this media but also quite a few issues opened here were actually questions (including this issue) so this may be a better option and foster the use of Github issues for real bugs and enhancements or feature requiests. We'll see...

gm8arv, 2017-12-22:
Changed the decimation to 8 giving a 375k sample rate and all is now well.  I hadn't appreciated the sampling rate required.  Perhaps the plug-in could give a warning?

Noted on the audio input being for TX - of course!

I think that the point about a group would be that everyone sees the messages, and perhaps could answer at least some of the questions rather than you having to answer everything, and leaving here for genuine bugs, and improving the signal-to-noise ratio here!  I've run several Yahoo groups and would be quite happy to help with setting up and running a new group for you if you like.

f4exb, 2017-12-22:
Glad all is well with DSD demod now. In fact all demods require at least 48 kS/s sampling rate and some even more so that it is adapted to required bandwidth. While it is fairly evident for wide FM and ATV I think the untold thing here is that all analog audio plugins and also DSD require 48 kS/s. For the audio plugins this is because the audio rate is fixed at 48 kHz (handled via portaudio). So indeed this should be made prominent in the plugins documentaiton.

I have started a group although not fully ready yet. I think I will mention it on the top of the readme in the next release.

closed 2017-12-31 by f4exb