f4exb/sdrangel #112: Enhancement: bass-cut filter for speech.

gm8arv, 2017-12-22:
Comparing the audio from D-Star on SDRangel and my IC-R8600 receiver, it's clear that the Icom box has significant low cut on the D-Star output, making it more like telephone quality, and rather easier to understand.  Would it be possible to add something like a 300 Hz high-pass filter option?  This could be a useful option for for all audio outputs.

f4exb, 2017-12-22:
It is already here and there. You have it for AM demodulation for example and it's good to have it as an option since you may want the bass for broadcast AM. In NFM it is mandatory and permanent due to the very frequent use of subtone squelch. 

So if it is an option for digital speech why not. In fact if you use the default mebelib soft codec the quality is really not as good as with a dongle from DVSI (which is also supported) and with the dongle it is no problem to have the bass frequencies it even makes it more natural (so to speak ...)

gm8arv, 2017-12-22:
Thanks!  Although I write software, and would love to help, my language is Delphi and my OS Windows, so not compatible in two ways.  Yes, it's the LF transients from the decoder, and general "make it telephone quality (at the low end) which will improve audibility.  An option, of course.

f4exb, 2017-12-23:
It turns out that it is a quite complex implementation. Because of the digital processing you only get access to audio samples by packets and these samples are short integers (S16LE) and there is no filter already available that work with integers.

gm8arv, 2017-12-23:
Thanks for checking.  I feel sure that there must be some code available somewhere to do a simple filtering job with integers?  All of my DSP work has been with integers (back when they were required for speed!).  A very simple filter would do, subtracting a low-pass (averaged) version of the signal from the original?

f4exb, 2017-12-26:
In fact it is just a matter of designing the IIR filter. Python scipy has filter design functions for this. So in fact this might not be as complex as I thought. For mbelib processing I would rather port the process in dsdcc.

gm8arv, 2017-12-26:
Oh, that's good news.  Something cutting below ~300 Hz would be fine, and I don't think it needs to be too sharp, so a relatively small length and little CPU load.  Sorry I can't help with the implementation, but you'll know best how to do that.

DG9BFC, 2017-12-30:
what about adding treble and bass sliders to audio out (and maybe also to audio in / mic in for tx)
or in short a tx and an rx equalizer

f4exb, 2017-12-30:
Linux users may use an external package like pulseaudio-equalizer since SDRangel interfaces with pulseaudio.

DG9BFC, 2017-12-30:
and what about the win users?? .... by the way ... i finally got the lime working with my old laptop (core2duo) ... can see spectrum and waterfall BUT no audio .. grmbl ... hopefully i can find out what i made wrong here in my setup :-( ...

f4exb, 2017-12-30:
I am sure there are options for Windows too but I am not a Windows specialist nor a fan. I simply don't  see the point of adding more complexity for audio processing inside the plugin.

gm8arv, 2017-12-31:
Perhaps this is making things more complex than needed?  A simple bass-cut in the audio output (Preferences, Audio, Audio output) path would be quite enough to remove many of the disturbing artefacts (and sounds like Icom have something similar in their IC-R8600 D-Star decoding).  If the cut-off is adjustable (even by a .INI file), even better.

f4exb, 2017-12-31:
I am sorry but I will not do this. This makes code excessively complex and inefficient. In the general case it has to be as an ugly hack on the audio FIFO.

f4exb, 2017-12-31:
Don in v3.9.0

DG9BFC, 2017-12-31:
i just wanted to give some ideas ... but if that simple low cut is enough ... ok on that
..... could anybody help me with my setup?? i have the same problems with rtl dongle that i have with the lime
when i click on plus sign (to ad a receiving filter in spectrum) the soft stops working

gm8arv, 2017-12-31:
Only tested on D-Star, but the cut seems enough to me, and a very welcome improvement.  If anything, the cut-off frequency could be a little lower, but not by much.  Many thanks for this enhancement!

f4exb, 2018-01-01:
I know it's a bit high because I just used the coefficients found in this document http://www.analog.com/media/en/technical-documentation/dsp-book/dsp_book_Ch20.pdf using the lowest cut-off possible (0.01) this is still 480 Hz. I was counting on the relatively slow roll-off with just a 2 pole filter. Maybe the Python scipy's signal package can help there. I was not successful with the bandpass but I know the lowpass and highpass should work.

f4exb, 2018-01-01:
I have created a discussion group that would be the right place to ask for help: https://groups.io/g/sdrangel

f4exb, 2018-01-03:
Cutoff frequency of high pass filter has been lowered in v3.9.1

gm8arv, 2018-01-03:
Excellent, but I wouldn't like to see it any lower than that.  Thanks for adding this helpful feature.

f4exb, 2018-01-03:
OK great

closed 2018-01-03 by f4exb