f4exb/sdrangel #115: no ATV modulator in the latest window versions 3.8.5  and 3.8.6

PE1PZW, 2017-12-29:
Hello f4sxb,
first off all , i want to give a compliment to you about the software.
But i miss in the latest versions the ATV modulator , why?
And is it possible to do DATV whith it in the future? (DATV-EXPRESS software?)

thanks and go on!!  fantastic!!

f4exb, 2017-12-29:
The DATV support is followed up in this issue: #18 Someone is already at it for the demodulator.
I don't know why the ATV modulator plugin is missing in the Windwos binary package. This is not intentional I will have to check for the next release.

f4exb, 2017-12-29:
I see modatv.dll in the plugins/channeltx directory of the archive. What makes you think the plugin is missing?

f4exb, 2017-12-30:
I didn't pay attention to the date but the modatv.dll is very old since it is not built due to a bug in MacOS specification (should be !macx and not macx). This probably explains because this old version should have a lot of unresolved references and therefore is rejected by the plugin loader.

f4exb, 2017-12-31:
Fixed in Windows 64 distribution of v3.9.0.
Note: this plugin is not available in the Windows 32 bit binary distribution.

closed 2017-12-31 by f4exb

PE1PZW, 2018-01-02:
thanks!!!

Closed #115<https://github.com/f4exb/sdrangel/issues/115>.