f4exb/sdrangel #117: AM Demodulator RF bandwidth setting in presets not being applied

brendanthebig, 2018-01-03:
When loading presets with AM Demodulator, I have noticed that the RF bandwidth setting is not being applied. It works ok when making manual adjustments. I have tried to re-save the preset after making bandwidth changes but problem still persists.

f4exb, 2018-01-04:
I also noticed the same with the SSB demod. It  may be due to changes in sample rate handling in which case it was introduced in 3.9.0

brendanthebig, 2018-01-04:
I haven't noticed the problem with SSB demod myself. AM demod had no issues in 3.9.0 for me.

f4exb, 2018-01-05:
There is definitely a problem with the sample rate used in the interpolator:
https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demodam/amdemod.cpp#L200

It was already there in 3.9.0:
https://github.com/f4exb/sdrangel/commit/d4767364874785c69e3e67100694fa4a4a13bf26

It also affects SSB demod and SSB mod in a similar way.

Fix is easy and will be part of next release.

brendanthebig, 2018-01-08:
Hi Edouard, V3.10.0 has fixed the issue with RF bandwidth on AM demod but there is still an issue with SSB demod. It seems that the cut-off frequency being applied is about half of what is selected.

f4exb, 2018-01-08:
The frequency shift has an influence on how many halfband decimators are used each stage halving the baseband sample rate before feeding the samples to the demod (this is the "channelizer" step).

f4exb, 2018-01-08:
There is an additional problem with the SSB demod filtering that I think has been there for ages. The demod decimator filter is set at half the required RF bandwidth. This is fine for double sideband signals when what is meant by RF bandwidth is the total double sideband bandwidth which is the case for all other demods. I suppose the decimator settings for the SSB were just copied and pasted and that was a long time ago...

brendanthebig, 2018-01-11:
Working well in V3.10.1...Good job Edouard...

closed 2018-01-11 by brendanthebig