f4exb/sdrangel #118: Crash when trying to press play button

LD723, 2018-01-04:
When I select the rtl dongle then press the green play button sdrangel crashes and I have higher than open gl 4. The graphics card I have is a gtx 1050 ti so it is pretty new. I can't figure out any fixes and I have tried multiple things but I still get the crash.

LD723, 2018-01-04:
Also forgot to mention I have windows 10 64 bit

f4exb, 2018-01-04:
Crashes are very difficult if at all possible to investigate on Windows unless the same situation is reproducible on Linux which is not the case here. What you may try to do:

- Activate logging to file with Preferences -> Logging then check "log to file" and select a file or leave the default. Restart the application, execute your scenario then the log file should have the debug messages until it crashes
- If you use the 64 bit version try the 32 bit version
- Run it on Linux either native or inside a Virtualbox VM

LD723, 2018-01-05:
Alrighty I looked at the debug messages and nothing looks like an error or unusual. I've also tried the 32 bit version but same problem and I can't try a virtualbox because there is very little space on my pc because some games on my computer take up a lot of hard drive space.

f4exb, 2018-01-18:
This question would be better answered at: https://groups.io/g/sdrangel

closed 2018-01-18 by f4exb