f4exb/sdrangel #120: No audio in DSD Demodulator

kallix, 2018-01-07:
Hi, 

I can't seem to obtain any audio out of the DSD Demodulator, even though the speaker icons are green (see screenshot)

I do have correct audio from NFM / WFM / Broadcast FM Demodulators. 

Am I missing something or doing something wrong ? 
Is there something special about how the DSD Demodulator handles sound ?

I'm running the current git head on Debian.

[image]

f4exb, 2018-01-07:
I guess you compile the software in which case you need to have mbelib installed in your system. If you have mbelib installed in a custom location, say /opt/install/mbelib you will need to add these defines to the cmake command: -DLIBMBE_INCLUDE_DIR=/opt/install/mbelib/include -DLIBMBE_LIBRARY=/opt/install/mbelib/lib/libmbe.so

f4exb, 2018-01-18:
This question would be better aswered at: https://groups.io/g/sdrangel

closed 2018-01-18 by f4exb

ivanenev, 2018-03-01:
Hi,
Any solution to this problem? 
I have the same issue although I compiled with mbelib support.

Thanks

sh123, 2018-11-21:
Same issue on debian, no audio output, while log indicates that VOX frames are received.

nukedclx, 2020-09-26:
same here, vox and speaker icon green and no audio..

vetrivelmnv13, 2020-11-25:
hi guys, i am trying to receive walkie-talkie signal by using my limesdrmini. the walkie- talkie doing 4FSK modulation. i am using sdr angel for reception. but sdrangel seems like, does not have any FSK demodulation. so i how can i receive the walkie talkie signal ? can anyone give me some suggestion

f4exb, 2020-11-25:
Question would be better answered here: https://groups.io/g/sdrangel

vetrivelmnv13, 2020-11-25:
> Question would be better answered here: https://groups.io/g/sdrangel

hi @f4exb ,
i am not able to post a question in sdrangel forum, can you guide me?