f4exb/sdrangel #125: Android apk

indzhev, 2018-01-17:
Hello. You can imagine how nice will be to have such app in android. Even simplified or for rooted devices only. There is nithing for dmr for android. Sdrangel can be the first one :)

f4exb, 2018-01-17:
Well Marty is already active on this. See this (long) topic: https://groups.io/g/sdrangel/topic/issue_when_compiling_sdrangel/7732033

I have tried to port the whole application a long time ago as Qt supposedly supports the Android toolchain. I could get an .apk out of it but unfortunately not working with what seemed to be complicated multi threading issues so I gave up.

Now there is another possibility and that's what Marty is at I think. Since we now have a server app with a REST API interface it is quite possible to build a client side Android app. Then we can see if it would be possible to port the server which would supposedly be simpler but neither Marty or myself have tried it. In any case a client side .apk is already well worth it.

f4exb, 2018-04-24:
Version after version Android gets more and more useless and becomes just a recreation environment not doing anything really useful. SDRangel will NEVER be supported in Android as it will not support it either. This issue can be closed.

f4exb, 2022-12-21:
Looks like thanks to @srcejon this one is back!

srcejon, 2022-12-21:
Now available on Google Play - https://play.google.com/store/apps/details?id=org.sdrangel&hl=en&gl=US

tianguo1314, 2024-05-21:
Hello, I am very interested in this Android project. May I ask if this SDRangel Android project is open source? Can you provide the SDRangel source code repository address for Android?

srcejon, 2024-05-21:
This is the repo for it.

tianguo1314, 2024-05-21:
What I need is a project that can run on AS (Android Studio). The current warehouse project should be a server-side one, right

srcejon, 2024-05-21:
There isn't one.

tianguo1314, 2024-05-21:
Can I understand that the Android version of SDRangel is a closed source project

srcejon, 2024-05-21:
It's built using cmake.

f4exb, 2024-07-04:
We may consider it implemented now

closed 2024-07-04 by f4exb