f4exb/sdrangel #126: Use RTP protocol for streaming

f4exb, 2018-01-18:
Today a few channel plugins use raw UDP to get audio samples out (in in one case). This is already good to connect to other software that are capable of treating raw samples like GNUradio. However this is not very convenient for a larger audience and the other end should have a priori knowledge to be able to interpret the samples correctly. RTP is widely used for streaming audio and video and could be a good alternative (raw UDP may stay as one option). I think the future client web or Android apps using SDRangel as a server could benefit from this.

With video one can imagine not only transmitting actual video from the ATV plugin but also the possibility to format the spectrum and scope views as video streams. In client/server mode this new feature could be very convenient.

F5OEO, 2018-01-18:
RTP is transporting generally H264 or transport stream. So just be aware you surely need a video codec or Video/Transport stream encapsulator. Feature is interesting as mobile phone has generally hardware video decoder( BaseLine profile) which allow to display without using CPU.

f4exb, 2018-01-26:
Possible C++ library: https://github.com/j0r1/JRTPLIB

f4exb, 2018-01-29:
RTP payload types: https://www.iana.org/assignments/rtp-parameters/rtp-parameters.xhtml#rtp-parameters-1

f4exb, 2018-01-29:
Alternative library: http://www.live555.com/liveMedia/
With Ubuntu/Debian packages

f4exb, 2018-01-30:
SDP for streaming 48 kHz audio 16 bit mono as dynamic payload (PT=96):
```
v=0
m=audio 0 RTP/AVP 96
a=rtpmap:96 L16/48000/1
```

F5OEO, 2018-01-30:
Seems that Live555 could be a good option. Gstreamer architecture is huge but think is a litlle overkill.

f4exb, 2018-01-30:
Well in fact JRTPlib is more low level but also seems easier to use. The session can be opened in the plugin just like with UDP. I think Live555 would require a global server instance in its own right like the HTTP server. This is possiblebut appears to be significantly more complex. For the time being we will just stream the audio out with RTP or UDP option.

F5OEO, 2018-01-30:
I bet before that you move to a low level lib, as soon as see, SDRAngel always get efficient and kiss solutions (appart maybe for opencv). Let's go on RTP audio but video with transport stream coming with DATV soon, I hope.

f4exb, 2018-02-05:
The concerned plugins are:
  - AM demod
  - Broadcast FM demod (stereo)
  - Digital voice (DSD) demod
  - Narrow FM demod
  - SSB demod
  - Wide FM demod

f4exb, 2018-02-22:
JRTPLIB does not fit into the Qt Network framework and is therefore not easily portable. However it has interesting features. In v3.12.0 I will deploy a Linux only solution applied to the NFM demod only. A more definite solution would be to rewrite a slimmed down version of JRTPLIB based on Qt Network retaining the essential features that are actually needed.

f4exb, 2018-03-09:
Have to remove DSD demod from the list so this ticket will be implemented in a reasonable amount of time. The problem with digital voice is that the audio does not come directly from the RF samples flow and therefore its actual rhythm is not stable. It makes it very difficult to use the RTP writer directly from the demod so this will have to be done differently.

f4exb, 2018-03-17:
Done in v3.13.0

closed 2018-03-17 by f4exb