f4exb/sdrangel #129: Building complains about boost library on osx

jbarea, 2018-01-26:
Hello all, just tried to build sdrangel on osx el capitan. I installed macports in order to install the other libraries and programs, such as hackrf libraries, gnuradio and the boost library. When attempting to build i get an error:
OSX make error:sdrangel/sdrbase/dsp/scopevis.h:4:10: fatal error: 'boost/circular_buffer.hpp' file not found
I saw that this error was referred in another issue and the user who reported the problem solved it installing boost library, the problem is i already have boost library installed.
I see that maybe is because it is stated as boost/pathtofile while i have the library in /opt/local/include/boost, as macports put it. Is there any place where i can specify the new path withouth the need to edit manually every file in the library?.
Thanks in advance!!!

f4exb, 2018-01-26:
There is no custom cmake find macro for Boost so it uses original cmake's. Looking at the documentation: https://cmake.org/cmake/help/v3.0/module/FindBoost.html The variables should be:

  - BOOST_INCLUDEDIR       - Preferred include directory e.g. <prefix>/include
  - BOOST_LIBRARYDIR       - Preferred library directory e.g. <prefix>/lib

You should be able to tweak the cmake command line using these variables

jbarea, 2018-01-27:
Thanks for the reply. I tried to add the variables at command_line using cmake -BOOST_INCLUDEDIR=/opt/local/include -BOOST_LIBRARYDIR=/opt/local/include/boost .., but still getting the same error. It doesn't find the boost library.

f4exb, 2018-01-27:
That's: `-DBOOST_INCLUDEDIR=/opt/local/include -DBOOST_LIBRARYDIR=/opt/local/include/boost`

jbarea, 2018-01-31:
Hello and thanks for all the help you are providing. I tried to use the modifiers as you specified, but still getting the same error on make. Boost library is installed via macports and i'm using the correct path. Still investigating what could be the prob.

f4exb, 2018-03-15:
This pull request from the same contributor has been merged recently: https://github.com/f4exb/sdrangel/pull/147

Do you still have issues?

f4exb, 2018-03-24:
Please use the discussion group: https://groups.io/g/sdrangel if the problem persists

closed 2018-03-24 by f4exb