f4exb/sdrangel #130: The forked libairspyhf (Airspy HF+) does not handle phase imbalance

touil, 2018-01-30:
Am I missing something or this code has no notion of phase imbalance?

https://github.com/f4exb/airspyhf/blob/intsamples/libairspyhf/src/iqconverter_int16.c

touil, 2018-01-30:
Also, the amplitude imbalance is only updated once for every buffer. This creates clicks between the buffers.

touil, 2018-01-30:
The NCO is quantized aggressively and generates spurs that mix with other signals within the passband.

f4exb, 2018-01-31:
1. You are probably missing this bit: ``` dest[i+1] = ((src[i] * cnv->imbalance) >> 16) - qCorr; ``` so there is indeed phase imbalance correction. This is the same algorithm than the one used for DC removal and IQ imbalance correction within SDRangel and it works. You can check for yourself with the test source.
2. Yes that is a possibility and an interesting point
3. The 16 bit samples are multiplied by 16 bit NCO samples then shifted back to 16 bits. You cannot make it better mor bits at the NCO is just a waste. Try for yourself: move the LO ppm slider so the NCO will be engaged then set it back to 0 with the "R" button and you will not notice any change other than the slight frequency move

touil, 2018-01-31:
1. Of course I read that part of the code. It only alters the DC offset and the relative amplitude. In no case it changes the relative phase between the I and Q branches. The term "imbalance" in your code only refers to the relative amplitude. This restricts the image rejection to 70 dBc. "It works" is very relative. For HF with narrow band signals, it's clearly not ebough. To fix that code, you need to:
a) Estimate the amplitude imbalance
b) Estimate the **phase imbalance**
c) Correct the amplitude imbalance according to the estimated **phase** adjustment
d) Estimate and correct the DC offset according to the phase and amplitude adjustment
e) Apply the adjustments: DC offset, phase and amplitude

Here's a paper to help you understand the underlying math: https://pdfs.semanticscholar.org/2ea9/145007038213abb2a1a6c794f4653701860b.pdf
Check equation(1). The term Q(t)*sin(φ) is **added** to the I(t) branch, and the amplitude of I(t) is adjusted to compensate for it. You will find the same formulas everywhere. Hint: at low values of φ, sin(φ) approaches φ.

2. Once implemented correctly, this part will be much less efficient than the original code using floats. We can also see the use of doubles mixed with fixed int16 operations and branching inside the processing loop. Did you check the generated assembly? You give no chance to the compiler to unroll this loop or optimize the whole arithmetics with SIMD.

3. Again, your assumption here is that the signals in question are wide band - except they are not. The 16bit NCO is at the edge for 10 kHz channels, but completely destroys the information in signals with a few Hz of bandwidth, like FT8, JT4, JT9, JT65, QRA64, ISCAT, etc. etc. The frequency resolution is another drawback that is not acceptable with such narrow band signals.

PS: Apart from the implementation problems, many assumptions are not applicable to the purpose for which the Airspy HF+ and the original code were designed. While I salute your efforts on this take, I remind you that DSP is all about correctness. You would go away with converting the properly processed samples in the original lib to whatever format your application needs.

f4exb, 2018-01-31:
Most of your comments are valid and well documented about the DC/IQ correction however I do not subscribe to the fixed point vs floatng point argument. 

There is nothing that can be done with floats that cannot be done with ints (fixed point) when the range of numbers you deal with is limited which is the case with DSP. Both are essentially the same the fixed point having an implicit fixed exponent that limits its range. I have seen a lot of discussions about performance of one vs the other and it is true that with modern processors having efficient FPUs the difference is usually not very relevant. There is much more to gain in vectorization and I agree that on that point this piece of code can be largely optimized. Also the mix of float and fixed can be removed referring to the very first statement of the paragraph. Yet I still have to see evidence that well written fixed point code perform worse than equally well written floating point code so the statement "this part will be much less efficient than the original code using floats" is just wrong. I take the challenge to prove it. Floats will provide some ease of use but if correctness is your concern as you mention you are more likely to make mistakes with floats as precision concerns may be overlooked.

Also I do not really understand the argument about the NCO. The 16 bit limitation is on the amplitude not the phase. There are 768000 digitized values so at a 768000 S/s sample rate you have 1 Hz precision (not 10k!). Of course because of symmertries only a fourth of this array is really needed but take it as a first approach. In the case of SDRangel and LO correction apart you don´t need this precision because the fine tuning is done at channel level. 

As I briefly mentioned the correction code written on my fork of airspyhf is directly derived from the code that can be found here: https://github.com/f4exb/sdrangel/blob/master/sdrbase/dsp/dspdevicesourceengine.cpp#L176 So all your valid remarks do apply here. To be honest this is very old code and I haven´t put a lot of attention into it until now. As charity begins at home I´d like to fix this first and I will change the title of the issue in that sense. Of course this has strong links with my fork of airspyhf and I hope this will be evident by the content of this issue. Indeed it seems not possible to open issues on forks.

It is also true that at present the test source only deals with amplitude I/Q imbalance which obviously hides some problems. So true phase imbalance has to be implemented there.

Also I'd like to clone the AirspyHF+ plugin into one that takes floats at its input from your original airspyhf library (yes!) in order to see the real difference in practice and have a base for comparison later. For that purpose I need to develop a floating point decimation chain so that conversion is on the low rate end. Thus it can also be used for real life performance comparison.

touil, 2018-01-31:
Fair enough. I'm not discussing fixed vs. float here. I have tons of code of both kinds.
For the NCO, try to check the spectral purity with a high resolution FFT. You will see spurs ~80 dBc lower. That's less than the DR of the receiver.
Concerning the IQ correction itself, a few years ago, I developed and documented the algorithm that is now implemented in gr-iqbal. It works fine for wide band signals, but to get 120 dBc image rejection I needed something better. That's basically the code you removed from your fork.

f4exb, 2018-02-04:
For SDRangel it will be enough to do frequency independent phase imbalance correction.

f4exb, 2018-02-04:
I tried both a fixed point and floating point fixed DC and frequency independent IQ imbalance correction on an average use. During about 7 minutes CPU time the global process CPU load averaged to one CPU:
  - floating point: 53.3 %
  - fixed point: 57 %  

Conclusion: the floating point vs fixed point debate is indeed a silly debate unless dealing with FPU less processors.

touil, 2018-02-04:
And since we only run on CPU's with hardware FP, why the hassle? Just use the floats and call it a day.

f4exb, 2018-02-05:
So the outcome is:
  - AirspyHF plugin will use the standard airspyhf library
  - DC+IQ imbalance in SDRangel will be fixed using a frequency independent phase correction. This already improves things a lot.
  - Converting the Rx DSP chain to floating point at least past the decimation is a possibility and this would remove the 16/24 bit choice at build time. This is a fairly large undertaking and is kept for later.

touil, 2018-02-05:
That sounds like a good plan. Have fun!

f4exb, 2018-02-23:
Corrections implemented in v3.12.0 in addition the original airspyhf library is used in Airspy HF plugin

closed 2018-02-23 by f4exb