f4exb/sdrangel #132: Channel window scrolls up and down while changing frequency

herrfeuer, 2018-01-31:
Hello, 
if i change Δf of a demodulator by mouse wheel the channel window scrolls up and down. 
This results in accidental changes of other demodulator frequencies if settings are expanded.
https://imgur.com/a/geseR
This behavior is only if you change frequency. If you change squelch by mouse wheel the demodulator
stays in place.

f4exb, 2018-01-31:
It is hard to tell by the link because the image is very tiny but if I get it right basically this is how the Qt GUI works. The thumb-wheels of the Δf are tied to the mouse wheel and I think this is quite convenient. So are other controls by default with Qt: sliders, combos. ... The slight drawback of this is that if you are just a bit off the control zone of the mouse events then the action is to scroll the parent frame.

When the mouse is in control of the thumb-wheels the pointer turns into a finger pointer and the background of the wheel gets slightly brighter so that should be a good enough indication.

herrfeuer, 2018-01-31:
I am sorry, but i could not record a video of showing the effect. So i made screenshots and a gif.
If i turn the mouse wheel above the thumb-wheels of the Δf the side scroller scrolls togehter and the whole window scrolls down. If i am over a slider the window doesn't scroll when i use the mousewheel.
Maybe this is a Qt bug.

herrfeuer, 2018-01-31:
https://youtu.be/1gQQ6BmjeZk

f4exb, 2018-01-31:
Thanks for the video now this is clear but strange also. Doing exactly the same thing works right for me. Which version of Qt is your build using and what system are you running on? This is displayed in the status bar at the bottom right of the main window.

herrfeuer, 2018-01-31:
SDRangel 3.11.1 Qt 5.7.1 64b x86_64 Debian GNU/Linux 9 (stretch)

herrfeuer, 2018-02-12:
It seemed QtWheelEvent was not proper terminated in valuedial and valuedialz Widgets.
Pull reqest #136 fixes that.

f4exb, 2018-02-23:
Implemented in v3.12.0

closed 2018-02-23 by f4exb