f4exb/sdrangel #133: LimeSDR Plugin - Failed build

FFY00, 2018-02-04:
I have the latest drivers from git.

```
[ 44%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/commandkeyreceiver.cpp.o
/home/anubis/git/sdrangel/plugins/samplesink/limesdroutput/limesdroutput.cpp: In member function ‘virtual bool LimeSDROutput::handleMessage(const Message&)’:
/home/anubis/git/sdrangel/plugins/samplesink/limesdroutput/limesdroutput.cpp:632:32: error: ‘struct lms_stream_status_t’ has no member named ‘sampleRate’
                         status.sampleRate,
                                ^~~~~~~~~~
[ 44%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/commandoutputdialog.cpp.o
/home/anubis/git/sdrangel/plugins/samplesource/limesdrinput/limesdrinput.cpp: In member function ‘virtual bool LimeSDRInput::handleMessage(const Message&)’:
/home/anubis/git/sdrangel/plugins/samplesource/limesdrinput/limesdrinput.cpp:631:32: error: ‘struct lms_stream_status_t’ has no member named ‘sampleRate’
                         status.sampleRate,
                                ^~~~~~~~~~
```

f4exb, 2018-02-04:
Obviously `lms_stream_status_t` is in the LimeSuite API amd good chances are that your copy or clone is outdated. See: http://docs.myriadrf.org/LMS_API/structlms__stream__status__t.html

f4exb, 2018-02-05:
In fact it is the opposite. sampleRate has been removed from the structure.

f4exb, 2018-02-05:
Removed on dev branch

FFY00, 2018-02-05:
Thanks, I'm able to build the `dev` branch now. Although I can only tune the LimeSDR between 3kHz and 30kHz. The main frequency is synchronized with the NCO frequency.

f4exb, 2018-02-05:
I don't  know what it is doing that. This is a regression of LimeSuite maybe try to check out an earlier commit. I suspect this was introduced with this one:
https://github.com/myriadrf/LimeSuite/commit/72e19bb054c46aed951eba306d0a8082bb87ebd1

f4exb, 2018-02-25:
Then you should be able to tune any frequency in the available range (30 to 3800 MHz)

FFY00, 2018-02-25:
When I tested it I wasn't able to. I'll try again when I have time.