f4exb/sdrangel #134: Ability to connect to SDRSpy server

Nocty201, 2018-02-07:
I love how well SDRAngel works with different demodulators compared to SDR#, but would love for it to be able to connect to a SpyServer ( https://airspy.com/spy-servers/ ) as it is more efficient with upstream bandwidth than sending the raw IQ data.

There is some discussion on third party clients here: https://airspy.groups.io/g/main/topic/6256611?p=,,,20,0,0,0::Created,,%23spyserver,20,2,20,6256611,ct=1&ct=1

Their remark: We tolerate other client implementations, but implementing servers or third party hardware support may be subject to special licensing.

f4exb, 2018-02-07:
Hello,

I don't think this is possible moreover if any kind of "special licensing" is involved. If you want to use a remote receiver or transmitter in conjunction with SDRangel as a client the companion project [SDRdaemon](https://github.com/f4exb/sdrdaemon) is made just for this purpose. The collection of supported hardware is a bit limited for the moment but my plans are to expand more on this rather than trying to deal with SpyServer. 

Also SDRangel comes now into GUI (sdrangel) and server (sdrangelsrv) flavors. I also plan to expand more on this to fit into distributed computing architectures.

Best regards.

Nocty201, 2018-02-07:
The way I read it, special licensing is for server side software as they "tolerate other client implementations". More than happy to approach them for clarification, but as I'm not a coder am unable to help with much more than that.

I'll have a look at SDRdaemon; the reason I'm using SpyServer at the moment is because the 5MB/s IQ stream can be reworked to send (in a lossier format) within 50KB/s of bandwidth. The latter works well from my consumer internet connection, the former not so much. Issue 14 of SDRdaemon makes me think it's raw IQ being sent, and I can't see anything saying otherwise (and if there is, forgive me for not spotting it).

Thanks!

f4exb, 2018-02-08:
Hello,

I was under the impression that SpyServer was not sending raw IQ but looking at the [code example](https://github.com/opensatelliteproject/xritdemod/blob/master/demodulator/src/SpyServerFrontend.cpp) in the post you mention it seems there are modes that do so (in various formats). It seems there is a FFT only mode that cannot be supported in SDRangel in ant way. So it may be possible to write a plugin to feed the IQ samples into the application but it looks like a fairly large undertaking and I'd rather direct my efforts about networking capabilities in the directions I mentioned at least in the near to middle future.

Best regards, Edouard.

f4exb, 2020-05-19:
Another implementation is here: https://github.com/racerxdl/gr-osmosdr/tree/master/lib/spyserver

tubalainen, 2021-12-10:
Second the spyserver feature. It would be faaaantastic!
I love the raw power of SDRangel - I simply cannot use it fully due to the fact that Ive got all my SDRs on the addict close(er) to my antennas due to the cable loss. They are all connected to a Asus Tinker Board 2. The USB sharing software's are glitchy at best and therefore the spyserver would be a great addition.

Thanks. 73ss de SM4XDJ

kevinelliott, 2022-01-23:
I would love SpyServer support as well!

f4exb, 2022-02-04:
@tubalainen I think that by "addict" you mean attic and so supposedly you run on a local network. Hence there are many other options including using SDRangel remote plugins:
  - [Remote sink](https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/remotesink/readme.md) at the distant spot
  - [Remote input](https://github.com/f4exb/sdrangel/blob/master/plugins/samplesource/remoteinput/readme.md) in the local instance of SDRangel. 

I am currently running SDRangel server flavor on RPi4 or RPi3B at the distant spot and it works very well. If you are on your own network there is no need to use an instence of Spyserver. Implementing a client Spyserver as a plugin in SDRangel is problematic but I have not ruled it out completely (that's why this issue is still open) yet I have no plans of implementing this in the short or mid term so you may consider the option just mentioned.

f4exb, 2022-02-10:
dropped eventually

closed 2022-02-10 by f4exb

f4exb, 2024-02-27:
Looks like we will eventually have it: https://github.com/f4exb/sdrangel/pull/2000