f4exb/sdrangel #137: Static analysis

mamborambo, 2018-02-20:
Hi @f4exb, first let me thank you for this wonderful project!
In my spare time I've ran a commercial static analyzer against sdrangel to see if it would uncover some bugs/problematic areas (which then could be better inspected to further improve this project). I've ran it against sdrangel's current HEAD (bfaf96f8ada7ed38e02651ec461790f3304880ac) after having built it on Ubuntu 16.04.3 amd64, here are the pre-build information from CMake:

```
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Boost version: 1.58.0
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.4
-- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found FFTW3F: /usr/include, /usr/lib/x86_64-linux-gnu/libfftw3f.so, /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so
-- Checking for module 'libdsdcc'
--   No package 'libdsdcc' found
-- libdsdcc not found.
-- Architecture: x86_64
-- Use SSE 4.1 SIMD instructions
-- Use SSSE3 SIMD instructions
-- Use SSE2 SIMD instructions
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.20
-- Found libusb-1.0: /usr/include/libusb-1.0, /usr/lib/x86_64-linux-gnu/libusb-1.0.so
-- Checking for module 'libbladeRF'
--   No package 'libbladeRF' found
-- libbladerf not found.
-- Checking for module 'libairspy'
--   No package 'libairspy' found
-- libhackrf not found.
-- Could not find Lime Suite
-- Checking for module 'libiio'
--   Found libiio, version 0.14
-- Found libiio: /usr/include, /usr/lib/x86_64-linux-gnu/libiio.so
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.5.3
-- Found librtlsdr: /usr/include, /usr/lib/x86_64-linux-gnu/librtlsdr.so
-- Checking for module 'libbladeRF'
--   No package 'libbladeRF' found
-- libbladerf not found.
-- Checking for module 'libairspy'
--   No package 'libairspy' found
-- libairspy not found.
-- Checking for module 'libairspyhf'
--   No package 'libairspyhf' found
-- libairspyhf not found.
-- Checking for module 'libairspy'
--   No package 'libairspy' found
-- libhackrf not found.
-- Checking for module 'libmirisdr'
--   No package 'libmirisdr' found
-- libmirisdr not found.
-- LibMiriSDR NOT found
-- Could not find Lime Suite
-- Checking for module 'libcm256cc'
--   No package 'libcm256cc' found
-- CM256cc not found
-- Could NOT find CM256CC (missing:  CM256CC_LIBRARIES CM256CC_INCLUDE_DIR) 
-- Checking for module 'libnanomsg'
--   Found libnanomsg, version 2.1.2
-- Found libnanomsg: /usr/include, /usr/lib/x86_64-linux-gnu/libnanomsg.so
-- Checking for module 'libbladeRF'
--   No package 'libbladeRF' found
-- libbladerf not found.
-- Checking for module 'libairspy'
--   No package 'libairspy' found
-- libhackrf not found.
-- Could not find Lime Suite
-- Checking for module 'libcm256cc'
--   No package 'libcm256cc' found
-- CM256cc not found
-- Could NOT find CM256CC (missing:  CM256CC_LIBRARIES CM256CC_INCLUDE_DIR) 
-- Checking for module 'libairspy'
--   No package 'libairspy' found
-- libhackrf not found.
-- Could not find Lime Suite
-- Checking for module 'libairspy'
--   No package 'libairspy' found
-- libhackrf not found.
-- Could not find Lime Suite
-- Configuring done
-- Generating done
-- Build files have been written to: sdrangel-master/build
```

The static analyzer has been configured to only emit high and medium severity/certainty messages. The report, a self-contained HTML, has been hosted [HERE](https://a.pomf.cat/nusegw.html). All the messages before located before the horizontal rule are classified as high severity/certainty ones, the remaining ones as medium severity/certainty.
Every static analyzer I know produces some false positives but I think some messages included in the report highlight true bugs/risky areas.

Cheers,
mamborambo

f4exb, 2018-02-21:
Many thanks! That's indeed very interesting and I will sure have a close look at it. So far I have only been able to make runtime checks using Valgrind and -fsanitize=address option on some scenarios thus not covering all cases. Static analysis can help a lot here.

mamborambo, 2018-02-22:
Thanks for the fixes!

> Static analysis can help a lot here.

I'm not sure but maybe you can try using the [Clang Static Analyzer](https://clang-analyzer.llvm.org/) (part of the Clang project) which is FOSS. :wink: 

I've ran the analysis again: this time against the `dev` branch (at 2b846f5d28da8f1186c43b478b92a5019fc994a8). I've also instructed CMake to build every module available (except `jrtplib`) in order to have that code sections checked too by the static analyzer:

```
$ cmake -DRX_SAMPLE_24BIT=ON -DLIBMIRISDR_LIBRARIES=/opt/install/libmirisdr/lib/libmirisdr.so -DLIBMIRISDR_INCLUDE_DIR=/opt/install/libmirisdr/include -DLIMESUITE_INCLUDE_DIR=/opt/install/LimeSuite/include -DLIMESUITE_LIBRARY=/opt/install/LimeSuite/lib/libLimeSuite.so -DLIBAIRSPYHF_LIBRARIES=/opt/install/libairspyhf/lib/libairspyhf.so -DLIBAIRSPYHF_INCLUDE_DIR=/opt/install/libairspyhf/include -DCM256CC_INCLUDE_DIR=/opt/install/cm256cc/include/cm256cc -DCM256CC_LIBRARIES=/opt/install/cm256cc/lib/libcm256cc.so -DLIBDSDCC_INCLUDE_DIR=/opt/install/dsdcc/include/dsdcc -DLIBDSDCC_LIBRARIES=/opt/install/dsdcc/lib/libdsdcc.so -DLIBSERIALDV_INCLUDE_DIR=/opt/install/serialdv/include/serialdv -DLIBSERIALDV_LIBRARY=/opt/install/serialdv/lib/libserialdv.so -DLIBMBE_INCLUDE_DIR=/opt/install/mbelib/include -DLIBMBE_LIBRARY=/opt/install/mbelib/lib/libmbe.so -DLIBPERSEUS_INCLUDE_DIR=/opt/install/libperseus/include -DLIBPERSEUS_LIBRARIES=/opt/install/libperseus/lib/libperseus-sdr.so ../
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Boost version: 1.58.0
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.4
-- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found FFTW3F: /usr/include, /usr/lib/x86_64-linux-gnu/libfftw3f.so, /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so
-- Checking for module 'jrtplib'
--   No package 'jrtplib' found
-- JRTPLib not found
-- Checking for module 'libdsdcc'
--   No package 'libdsdcc' found
-- Found libdsdcc: /opt/install/dsdcc/include/dsdcc, /opt/install/dsdcc/lib/libdsdcc.so
-- Found LibMbe: /opt/install/mbelib/include, /opt/install/mbelib/lib/libmbe.so
-- Found libserialdv: /opt/install/serialdv/include/serialdv, /opt/install/serialdv/lib/libserialdv.so
-- Architecture: x86_64
-- Use SSE 4.1 SIMD instructions
-- Use SSSE3 SIMD instructions
-- Use SSE2 SIMD instructions
-- Compiling for 24 bit Rx DSP chain
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.20
-- Found libusb-1.0: /usr/include/libusb-1.0, /usr/lib/x86_64-linux-gnu/libusb-1.0.so
-- Checking for module 'libbladeRF'
--   Found libbladeRF, version 1.5.1-0.2016.01~rc1-3
-- Found libbladerf: /usr/include, /usr/lib/x86_64-linux-gnu/libbladeRF.so
-- Checking for module 'libairspy'
--   Found libairspy, version 1.0
-- Found libhackrf: /usr/include, /usr/lib/x86_64-linux-gnu/libhackrf.so
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Checking for module 'libiio'
--   Found libiio, version 0.14
-- Found libiio: /usr/include, /usr/lib/x86_64-linux-gnu/libiio.so
-- Checking for module 'libperseus'
--   No package 'libperseus' found
-- Found libperseus: /opt/install/libperseus/include, /opt/install/libperseus/lib/libperseus-sdr.so
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.5.3
-- Found librtlsdr: /usr/include, /usr/lib/x86_64-linux-gnu/librtlsdr.so
-- Checking for module 'libairspy'
--   Found libairspy, version 1.0
-- Found libairspy: /usr/include, /usr/lib/x86_64-linux-gnu/libairspy.so
-- Checking for module 'libairspyhf'
--   No package 'libairspyhf' found
-- Found libairspyhf: /opt/install/libairspyhf/include, /opt/install/libairspyhf/lib/libairspyhf.so
-- Checking for module 'libmirisdr'
--   No package 'libmirisdr' found
-- Found libmirisdr: /opt/install/libmirisdr/include, /opt/install/libmirisdr/lib/libmirisdr.so
-- LibMiriSDR found
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Checking for module 'libcm256cc'
--   No package 'libcm256cc' found
-- Found CM256cc: /opt/install/cm256cc/include/cm256cc, /opt/install/cm256cc/lib/libcm256cc.so
-- Found CM256CC: /opt/install/cm256cc/lib/libcm256cc.so  
-- Checking for module 'libnanomsg'
--   Found libnanomsg, version 2.1.2
-- Found libnanomsg: /usr/include, /usr/lib/x86_64-linux-gnu/libnanomsg.so
-- SDRdaemonSource: use SSSE3 SIMD
-- Add Persesus plugin
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Checking for module 'libcm256cc'
--   No package 'libcm256cc' found
-- Found CM256cc: /opt/install/cm256cc/include/cm256cc, /opt/install/cm256cc/lib/libcm256cc.so
-- SDRdaemonFEC: use SSSE3 SIMD
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Configuring done
-- Generating done
-- Build files have been written to: sdrangel-dev/build
```
You can find the generated HTML report: [HERE](https://a.pomf.cat/kzdjmg.html).

Cheers,
mamborambo

f4exb, 2018-02-24:
It seems clang cannot get the issues PVS sees:
```
scan-build: Removing directory '/tmp/scan-build-2018-02-24-105805-15979-1' because it contains no reports.
scan-build: No bugs found
```

mamborambo, 2018-03-05:
As lots of changes have been introduced in the `dev` branch, let me know if you'd like to inspect another static analysis report before merging those to `master` and releasing version 3.13.0 (in that case I think it's better to proceed only after all sizable changes have been committed).

Cheers,
mamborambo

f4exb, 2018-03-05:
Sure, let's wait a bit. I am not ready yet to merge.

f4exb, 2018-03-17:
Now v3.13.0 is delivered so either master or dev has latest code

mamborambo, 2018-04-03:
Hi @f4exb, I've ran the analysis against the latest tagged version ([v3.14.1](https://github.com/f4exb/sdrangel/releases/tag/v3.14.1)). Hope you'll find it useful.

```
$ cmake -DRX_SAMPLE_24BIT=ON -DLIBMIRISDR_LIBRARIES=/opt/install/libmirisdr/lib/libmirisdr.so -DLIBMIRISDR_INCLUDE_DIR=/opt/install/libmirisdr/include -DLIMESUITE_INCLUDE_DIR=/opt/install/LimeSuite/include -DLIMESUITE_LIBRARY=/opt/install/LimeSuite/lib/libLimeSuite.so -DLIBAIRSPYHF_LIBRARIES=/opt/install/libairspyhf/lib/libairspyhf.so -DLIBAIRSPYHF_INCLUDE_DIR=/opt/install/libairspyhf/include -DCM256CC_INCLUDE_DIR=/opt/install/cm256cc/include/cm256cc -DCM256CC_LIBRARIES=/opt/install/cm256cc/lib/libcm256cc.so -DLIBDSDCC_INCLUDE_DIR=/opt/install/dsdcc/include/dsdcc -DLIBDSDCC_LIBRARIES=/opt/install/dsdcc/lib/libdsdcc.so -DLIBSERIALDV_INCLUDE_DIR=/opt/install/serialdv/include/serialdv -DLIBSERIALDV_LIBRARY=/opt/install/serialdv/lib/libserialdv.so -DLIBMBE_INCLUDE_DIR=/opt/install/mbelib/include -DLIBMBE_LIBRARY=/opt/install/mbelib/lib/libmbe.so -DLIBPERSEUS_INCLUDE_DIR=/opt/install/libperseus/include -DLIBPERSEUS_LIBRARIES=/opt/install/libperseus/lib/libperseus-sdr.so ../
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Boost version: 1.58.0
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.4
-- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found FFTW3F: /usr/include, /usr/lib/x86_64-linux-gnu/libfftw3f.so, /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so
-- Checking for module 'libdsdcc'
--   No package 'libdsdcc' found
-- Found libdsdcc: /opt/install/dsdcc/include/dsdcc, /opt/install/dsdcc/lib/libdsdcc.so
-- Found LibMbe: /opt/install/mbelib/include, /opt/install/mbelib/lib/libmbe.so
-- Found libserialdv: /opt/install/serialdv/include/serialdv, /opt/install/serialdv/lib/libserialdv.so
-- Architecture: x86_64
-- Use SSE 4.1 SIMD instructions
-- Use SSSE3 SIMD instructions
-- Use SSE2 SIMD instructions
-- Compiling for 24 bit Rx DSP chain
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.20
-- Found libusb-1.0: /usr/include/libusb-1.0, /usr/lib/x86_64-linux-gnu/libusb-1.0.so
-- Checking for module 'libbladeRF'
--   Found libbladeRF, version 1.5.1-0.2016.01~rc1-3
-- Found libbladerf: /usr/include, /usr/lib/x86_64-linux-gnu/libbladeRF.so
-- Checking for module 'libhackrf'
--   Found libhackrf, version 0.4
-- Found libhackrf: /usr/include, /usr/lib/x86_64-linux-gnu/libhackrf.so
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Checking for module 'libiio'
--   Found libiio, version 0.14
-- Found libiio: /usr/include, /usr/lib/x86_64-linux-gnu/libiio.so
-- Checking for module 'libperseus'
--   No package 'libperseus' found
-- Found libperseus: /opt/install/libperseus/include, /opt/install/libperseus/lib/libperseus-sdr.so
-- Checking for module 'libavcodec'
--   Found libavcodec, version 56.60.100
-- Checking for module 'libavformat'
--   Found libavformat, version 56.40.101
-- Checking for module 'libavdevice'
--   No package 'libavdevice' found
-- Checking for module 'libavutil'
--   Found libavutil, version 54.31.100
-- Checking for module 'libavfilter'
--   No package 'libavfilter' found
-- Checking for module 'libswscale'
--   Found libswscale, version 3.1.101
-- Checking for module 'libpostproc'
--   No package 'libpostproc' found
-- Checking for module 'libswresample'
--   Found libswresample, version 1.2.101
-- Found FFmpeg: /usr/lib/x86_64-linux-gnu/libavcodec.so;/usr/lib/x86_64-linux-gnu/libavformat.so;/usr/lib/x86_64-linux-gnu/libavutil.so  
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.5.3
[... 12 lines cut ...]
-- Checking for module 'libcm256cc'
--   No package 'libcm256cc' found
-- Found CM256cc: /opt/install/cm256cc/include/cm256cc, /opt/install/cm256cc/lib/libcm256cc.so
-- Found CM256CC: /opt/install/cm256cc/lib/libcm256cc.so  
-- Checking for module 'libnanomsg'
--   Found libnanomsg, version 2.1.2
-- Found libnanomsg: /usr/include, /usr/lib/x86_64-linux-gnu/libnanomsg.so
-- SDRdaemonSource: use SSSE3 SIMD
-- Add Persesus plugin
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Checking for module 'libcm256cc'
--   No package 'libcm256cc' found
-- Found CM256cc: /opt/install/cm256cc/include/cm256cc, /opt/install/cm256cc/lib/libcm256cc.so
-- SDRdaemonFEC: use SSSE3 SIMD
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Configuring done
-- Generating done
-- Build files have been written to: sdrangel-3.14.1/build
```

The generated HTML report can be found [HERE](https://a.pomf.cat/nokokb.html).

Cheers,
mamborambo

f4exb, 2018-04-03:
Many thanks! I will sure have a close look.
Best regards, Edouard.

mamborambo, 2018-06-20:
Hi @f4exb, I've ran the analysis against the latest major version ([v4.0.0](https://github.com/f4exb/sdrangel/releases/tag/v4.0.0)). Hope you'll find it useful.
The generated (zipped) HTML report can be found [HERE](https://github.com/f4exb/sdrangel/files/2120779/sdrangel_v4.0.0_static.zip).

Cheers,
mamborambo

mamborambo, 2018-06-20:
Unrelated to this issue but it seems I can't compile 1eb8716a9f70ed0253bb65dafae55025649be8af (tested on the same system which just compiled [v4.0.0](https://github.com/f4exb/sdrangel/releases/tag/v4.0.0); same cmake invocation): I don't know if it's expected...

```
$ cmake -DRX_SAMPLE_24BIT=ON -DLIBMIRISDR_LIBRARIES=/opt/install/libmirisdr/lib/libmirisdr.so -DLIBMIRISDR_INCLUDE_DIR=/opt/install/libmirisdr/include -DLIMESUITE_INCLUDE_DIR=/opt/install/LimeSuite/include -DLIMESUITE_LIBRARY=/opt/install/LimeSuite/lib/libLimeSuite.so -DLIBAIRSPYHF_LIBRARIES=/opt/install/libairspyhf/lib/libairspyhf.so -DLIBAIRSPYHF_INCLUDE_DIR=/opt/install/libairspyhf/include -DCM256CC_INCLUDE_DIR=/opt/install/cm256cc/include/cm256cc -DCM256CC_LIBRARIES=/opt/install/cm256cc/lib/libcm256cc.so -DLIBDSDCC_INCLUDE_DIR=/opt/install/dsdcc/include/dsdcc -DLIBDSDCC_LIBRARIES=/opt/install/dsdcc/lib/libdsdcc.so -DLIBSERIALDV_INCLUDE_DIR=/opt/install/serialdv/include/serialdv -DLIBSERIALDV_LIBRARY=/opt/install/serialdv/lib/libserialdv.so -DLIBMBE_INCLUDE_DIR=/opt/install/mbelib/include -DLIBMBE_LIBRARY=/opt/install/mbelib/lib/libmbe.so -DLIBPERSEUS_INCLUDE_DIR=/opt/install/libperseus/include -DLIBPERSEUS_LIBRARIES=/opt/install/libperseus/lib/libperseus-sdr.so ../
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Boost version: 1.58.0
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.4
-- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found FFTW3F: /usr/include, /usr/lib/x86_64-linux-gnu/libfftw3f.so, /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so
-- Checking for module 'libdsdcc'
--   No package 'libdsdcc' found
-- Found libdsdcc: /opt/install/dsdcc/include/dsdcc, /opt/install/dsdcc/lib/libdsdcc.so
-- Found LibMbe: /opt/install/mbelib/include, /opt/install/mbelib/lib/libmbe.so
-- Found libserialdv: /opt/install/serialdv/include/serialdv, /opt/install/serialdv/lib/libserialdv.so
-- Architecture: x86_64
-- Use SSE 4.1 SIMD instructions
-- Use SSSE3 SIMD instructions
-- Use SSE2 SIMD instructions
-- Compiling for 24 bit Rx DSP chain
-- Compiling with SIMD instructions for DSP if available
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.20
-- Found libusb-1.0: /usr/include/libusb-1.0, /usr/lib/x86_64-linux-gnu/libusb-1.0.so
-- Checking for module 'libbladeRF'
--   Found libbladeRF, version 1.5.1-0.2016.01~rc1-3
-- Found libbladerf: /usr/include, /usr/lib/x86_64-linux-gnu/libbladeRF.so
-- Checking for module 'libhackrf'
--   Found libhackrf, version 0.4
-- Found libhackrf: /usr/include, /usr/lib/x86_64-linux-gnu/libhackrf.so
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Checking for module 'libiio'
--   Found libiio, version 0.15
-- Found libiio: /usr/include, /usr/lib/x86_64-linux-gnu/libiio.so
-- Checking for module 'libperseus'
--   No package 'libperseus' found
-- Found libperseus: /opt/install/libperseus/include, /opt/install/libperseus/lib/libperseus-sdr.so
-- Checking for module 'libavcodec'
--   Found libavcodec, version 56.60.100
-- Checking for module 'libavformat'
--   Found libavformat, version 56.40.101
-- Checking for module 'libavdevice'
--   No package 'libavdevice' found
-- Checking for module 'libavutil'
--   Found libavutil, version 54.31.100
-- Checking for module 'libavfilter'
--   No package 'libavfilter' found
-- Checking for module 'libswscale'
--   Found libswscale, version 3.1.101
-- Checking for module 'libpostproc'
--   No package 'libpostproc' found
-- Checking for module 'libswresample'
--   Found libswresample, version 1.2.101
-- Found FFmpeg: /usr/lib/x86_64-linux-gnu/libavcodec.so;/usr/lib/x86_64-linux-gnu/libavformat.so;/usr/lib/x86_64-linux-gnu/libavutil.so  
-- Checking for module 'libairspy'
[... 737 lines cut ...]
[ 48%] Built target demoddatv_automoc
[ 48%] Generating ui_datvdemodgui.h
Scanning dependencies of target demoddatv
[ 48%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvdemod.cpp.o
[ 48%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvdemodgui.cpp.o
[ 48%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvdemodplugin.cpp.o
[ 48%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvideostream.cpp.o
[ 48%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvideorender.cpp.o
sdrangel-dev/plugins/channelrx/demoddatv/datvideorender.cpp: In member function ‘int DATVideoRender::new_decode(AVCodecContext*, AVFrame*, int*, AVPacket*)’:
sdrangel-dev/plugins/channelrx/demoddatv/datvideorender.cpp:616:45: error: ‘avcodec_send_packet’ was not declared in this scope
         ret = avcodec_send_packet(avctx, pkt);
                                             ^
[... cut ...]
plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/build.make:162: recipe for target 'plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvideorender.cpp.o' failed
make[2]: *** [plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvideorender.cpp.o] Error 1
CMakeFiles/Makefile2:2613: recipe for target 'plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/all' failed
make[1]: *** [plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

Cheers,
mamborambo

f4exb, 2018-06-21:
Thanks for the analysis I will have a look before next release. I made some changes in 4.0.0 using ffmpeg v 3.4. Systems with a ffmpeg version lower that 3.2 will fail. ffmpeg is pretty bad at upward compatibility. In the next release I plan to check for ffmpeg 3.2+ and skip the DATV plugin if not present. You won't get DATV but it will compile and not a lot of people (yet) are interested in DATV. Accordingly the xenial package will have it missing also.

mamborambo, 2018-11-16:
Hi, I've just tried to compile sdrangel v4.2.4 on Ubuntu 16.04.5 amd64 without success (wanted to run static analysis again): is this caused by an issue on my end?

```make
$ cmake -DRX_SAMPLE_24BIT=ON -DLIBMIRISDR_LIBRARIES=/opt/install/libmirisdr/lib/libmirisdr.so -DLIBMIRISDR_INCLUDE_DIR=/opt/install/libmirisdr/include -DLIMESUITE_INCLUDE_DIR=/opt/install/LimeSuite/include -DLIMESUITE_LIBRARY=/opt/install/LimeSuite/lib/libLimeSuite.so -DLIBAIRSPYHF_LIBRARIES=/opt/install/libairspyhf/lib/libairspyhf.so -DLIBAIRSPYHF_INCLUDE_DIR=/opt/install/libairspyhf/include -DCM256CC_INCLUDE_DIR=/opt/install/cm256cc/include/cm256cc -DCM256CC_LIBRARIES=/opt/install/cm256cc/lib/libcm256cc.so -DLIBDSDCC_INCLUDE_DIR=/opt/install/dsdcc/include/dsdcc -DLIBDSDCC_LIBRARIES=/opt/install/dsdcc/lib/libdsdcc.so -DLIBSERIALDV_INCLUDE_DIR=/opt/install/serialdv/include/serialdv -DLIBSERIALDV_LIBRARY=/opt/install/serialdv/lib/libserialdv.so -DLIBMBE_INCLUDE_DIR=/opt/install/mbelib/include -DLIBMBE_LIBRARY=/opt/install/mbelib/lib/libmbe.so -DLIBPERSEUS_INCLUDE_DIR=/opt/install/libperseus/include -DLIBPERSEUS_LIBRARIES=/opt/install/libperseus/lib/libperseus-sdr.so -DLIBBLADERF_LIBRARIES=/opt/install/libbladeRF/lib/libbladeRF.so -DLIBBLADERF_INCLUDE_DIR=/opt/install/libbladeRF/include ../
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Boost version: 1.58.0
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.4
-- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found FFTW3F: /usr/include, /usr/lib/x86_64-linux-gnu/libfftw3f.so, /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so
-- Checking for module 'libdsdcc'
--   No package 'libdsdcc' found
-- Found libdsdcc: /opt/install/dsdcc/include/dsdcc, /opt/install/dsdcc/lib/libdsdcc.so
-- Found LibMbe: /opt/install/mbelib/include, /opt/install/mbelib/lib/libmbe.so
-- Found libserialdv: /opt/install/serialdv/include/serialdv, /opt/install/serialdv/lib/libserialdv.so
-- Architecture: x86_64
-- Use SSE 4.1 SIMD instructions
-- Use SSSE3 SIMD instructions
-- Use SSE2 SIMD instructions
-- Compiling for 24 bit Rx DSP chain
-- Compiling with SIMD instructions for DSP if available
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.20
-- Found libusb-1.0: /usr/include/libusb-1.0, /usr/lib/x86_64-linux-gnu/libusb-1.0.so
-- Checking for module 'libbladeRF'
--   No package 'libbladeRF' found
-- Found libbladerf: /opt/install/libbladeRF/include, /opt/install/libbladeRF/lib/libbladeRF.so
-- Checking for module 'libhackrf'
--   Found libhackrf, version 0.4
-- Found libhackrf: /usr/include, /usr/lib/x86_64-linux-gnu/libhackrf.so
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Checking for module 'libiio'
--   Found libiio, version 0.15
-- Found libiio: /usr/include, /usr/lib/x86_64-linux-gnu/libiio.so
-- Checking for module 'libperseus'
--   No package 'libperseus' found
-- Found libperseus: /opt/install/libperseus/include, /opt/install/libperseus/lib/libperseus-sdr.so
-- Checking for module 'libavcodec'
--   Found libavcodec, version 57.107.100
-- Checking for module 'libavformat'
--   Found libavformat, version 57.83.100
-- Checking for module 'libavdevice'
--   No package 'libavdevice' found
-- Checking for module 'libavutil'
--   Found libavutil, version 55.78.100
-- Checking for module 'libavfilter'
--   No package 'libavfilter' found
-- Checking for module 'libswscale'
--   Found libswscale, version 4.8.100
-- Checking for module 'libpostproc'
--   No package 'libpostproc' found
-- Checking for module 'libswresample'
--   Found libswresample, version 2.9.100
-- Found FFmpeg: /usr/lib/x86_64-linux-gnu/libavcodec.so;/usr/lib/x86_64-linux-gnu/libavformat.so;/usr/lib/x86_64-linux-gnu/libavutil.so  
-- FFmpeg version 3.4.4-1~16.04.york0 found
[... 842 lines cut ...]
                 from ./sdrangel-4.2.4/plugins/channelrx/daemonsink/daemonsinkthread.h:28,
                 from ./sdrangel-4.2.4/plugins/channelrx/daemonsink/daemonsink.cpp:35:
/opt/install/cm256cc/include/cm256cc/gf256.h: In function ‘uint8_t gf256_mul(uint8_t, uint8_t)’:
/opt/install/cm256cc/include/cm256cc/gf256.h:154:16: error: ‘GF256_MUL_TABLE’ was not declared in this scope
         return GF256_MUL_TABLE[((unsigned)y << 8) + x];
                ^
/opt/install/cm256cc/include/cm256cc/gf256.h: In function ‘uint8_t gf256_div(uint8_t, uint8_t)’:
/opt/install/cm256cc/include/cm256cc/gf256.h:161:16: error: ‘GF256_DIV_TABLE’ was not declared in this scope
         return GF256_DIV_TABLE[((unsigned)y << 8) + x];
                ^
/opt/install/cm256cc/include/cm256cc/gf256.h: In function ‘uint8_t gf256_inv(uint8_t)’:
/opt/install/cm256cc/include/cm256cc/gf256.h:167:16: error: ‘GF256_INV_TABLE’ was not declared in this scope
         return GF256_INV_TABLE[x];
                ^
[... cut ...]
plugins/channelrx/daemonsink/CMakeFiles/daemonsink.dir/build.make:66: recipe for target 'plugins/channelrx/daemonsink/CMakeFiles/daemonsink.dir/daemonsink.cpp.o' failed
make[2]: *** [plugins/channelrx/daemonsink/CMakeFiles/daemonsink.dir/daemonsink.cpp.o] Error 1
CMakeFiles/Makefile2:2805: recipe for target 'plugins/channelrx/daemonsink/CMakeFiles/daemonsink.dir/all' failed
make[1]: *** [plugins/channelrx/daemonsink/CMakeFiles/daemonsink.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

Cheers,
mamborambo

f4exb, 2018-11-17:
My bad: export.h is not installed by cm256cc cmake file. This is fixed in latest version of cm256cc (v1.0.4) so you can `git pull` it, recompile and reinstall cm256cc and it should work.

mamborambo, 2018-11-17:
Hi @f4exb, I've ran the analysis against the latest tagged version ([v4.2.4](https://github.com/f4exb/sdrangel/releases/tag/v4.2.4)). Hope you'll find it useful.

```make
$ cmake -DRX_SAMPLE_24BIT=ON -DLIBMIRISDR_LIBRARIES=/opt/install/libmirisdr/lib/libmirisdr.so -DLIBMIRISDR_INCLUDE_DIR=/opt/install/libmirisdr/include -DLIMESUITE_INCLUDE_DIR=/opt/install/LimeSuite/include -DLIMESUITE_LIBRARY=/opt/install/LimeSuite/lib/libLimeSuite.so -DLIBAIRSPYHF_LIBRARIES=/opt/install/libairspyhf/lib/libairspyhf.so -DLIBAIRSPYHF_INCLUDE_DIR=/opt/install/libairspyhf/include -DCM256CC_INCLUDE_DIR=/opt/install/cm256cc/include/cm256cc -DCM256CC_LIBRARIES=/opt/install/cm256cc/lib/libcm256cc.so -DLIBDSDCC_INCLUDE_DIR=/opt/install/dsdcc/include/dsdcc -DLIBDSDCC_LIBRARIES=/opt/install/dsdcc/lib/libdsdcc.so -DLIBSERIALDV_INCLUDE_DIR=/opt/install/serialdv/include/serialdv -DLIBSERIALDV_LIBRARY=/opt/install/serialdv/lib/libserialdv.so -DLIBMBE_INCLUDE_DIR=/opt/install/mbelib/include -DLIBMBE_LIBRARY=/opt/install/mbelib/lib/libmbe.so -DLIBPERSEUS_INCLUDE_DIR=/opt/install/libperseus/include -DLIBPERSEUS_LIBRARIES=/opt/install/libperseus/lib/libperseus-sdr.so -DLIBBLADERF_LIBRARIES=/opt/install/libbladeRF/lib/libbladeRF.so -DLIBBLADERF_INCLUDE_DIR=/opt/install/libbladeRF/include ../
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Boost version: 1.58.0
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.4
-- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found FFTW3F: /usr/include, /usr/lib/x86_64-linux-gnu/libfftw3f.so, /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so
-- Checking for module 'libdsdcc'
--   No package 'libdsdcc' found
-- Found libdsdcc: /opt/install/dsdcc/include/dsdcc, /opt/install/dsdcc/lib/libdsdcc.so
-- Found LibMbe: /opt/install/mbelib/include, /opt/install/mbelib/lib/libmbe.so
-- Found libserialdv: /opt/install/serialdv/include/serialdv, /opt/install/serialdv/lib/libserialdv.so
-- Architecture: x86_64
-- Use SSE 4.1 SIMD instructions
-- Use SSSE3 SIMD instructions
-- Use SSE2 SIMD instructions
-- Compiling for 24 bit Rx DSP chain
-- Compiling with SIMD instructions for DSP if available
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.20
-- Found libusb-1.0: /usr/include/libusb-1.0, /usr/lib/x86_64-linux-gnu/libusb-1.0.so
-- Checking for module 'libbladeRF'
--   No package 'libbladeRF' found
-- Found libbladerf: /opt/install/libbladeRF/include, /opt/install/libbladeRF/lib/libbladeRF.so
-- Checking for module 'libhackrf'
--   Found libhackrf, version 0.4
-- Found libhackrf: /usr/include, /usr/lib/x86_64-linux-gnu/libhackrf.so
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Checking for module 'libiio'
--   Found libiio, version 0.15
-- Found libiio: /usr/include, /usr/lib/x86_64-linux-gnu/libiio.so
-- Checking for module 'libperseus'
--   No package 'libperseus' found
-- Found libperseus: /opt/install/libperseus/include, /opt/install/libperseus/lib/libperseus-sdr.so
-- Checking for module 'libavcodec'
--   Found libavcodec, version 57.107.100
-- Checking for module 'libavformat'
--   Found libavformat, version 57.83.100
-- Checking for module 'libavdevice'
--   No package 'libavdevice' found
-- Checking for module 'libavutil'
--   Found libavutil, version 55.78.100
-- Checking for module 'libavfilter'
--   No package 'libavfilter' found
-- Checking for module 'libswscale'
--   Found libswscale, version 4.8.100
-- Checking for module 'libpostproc'
--   No package 'libpostproc' found
-- Checking for module 'libswresample'
--   Found libswresample, version 2.9.100
-- Found FFmpeg: /usr/lib/x86_64-linux-gnu/libavcodec.so;/usr/lib/x86_64-linux-gnu/libavformat.so;/usr/lib/x86_64-linux-gnu/libavutil.so  
-- FFmpeg version 3.4.4-1~16.04.york0 found
[... 37 lines cut ...]
-- Checking for module 'libcm256cc'
--   No package 'libcm256cc' found
-- Found CM256cc: /opt/install/cm256cc/include/cm256cc, /opt/install/cm256cc/lib/libcm256cc.so
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Checking for module 'libcm256cc'
--   No package 'libcm256cc' found
-- Found CM256cc: /opt/install/cm256cc/include/cm256cc, /opt/install/cm256cc/lib/libcm256cc.so
-- SDRdaemonFEC: use SSSE3 SIMD
-- Found Lime Suite: /opt/install/LimeSuite/include, /opt/install/LimeSuite/lib/libLimeSuite.so
-- Server: add Persesus plugin
-- Checking for module 'libcm256cc'
--   No package 'libcm256cc' found
-- Found CM256cc: /opt/install/cm256cc/include/cm256cc, /opt/install/cm256cc/lib/libcm256cc.so
-- SDRdaemonSource: use SSSE3 SIMD
-- LibMiriSDR found
-- Configuring done
-- Generating done
-- Build files have been written to: sdrangel-4.2.4/build
```

The generated (zipped) HTML report can be found [HERE](https://github.com/f4exb/sdrangel/files/2592072/sdrangel_v4.2.4_static.html.zip
).

Cheers,
mamborambo

f4exb, 2018-11-19:
Sure! that's always very useful and fixes are going to enter the next revision.

mamborambo, 2019-06-12:
Hi @f4exb, I've run the analysis against the latest tagged version ([v4.10.0](https://github.com/f4exb/sdrangel/releases/tag/v4.10.0)). Hope you'll find it useful.
I've compiled sdrangel from its source code with all the plugins (on Ubuntu 18.04.2 - as documented in the Wiki).
The generated (zipped) HTML report can be found [HERE](https://github.com/f4exb/sdrangel/files/3282811/sdrangel_v4.10.0_static.zip).

Cheers,
mamborambo

f4exb, 2019-06-14:
It is certainly useful. Many thanks!