f4exb/sdrangel #139: Unable to tune to frequency

sgoudelis, 2018-02-24:
Hello,

I just compiled SdrAngel with LimeSDR mini and RTLSDR support. Everything compliled fine. However, when using the Limesdr mini I cannot tune to any frequency (for RX). It will only allow me to tune to 30,000khz and 3,800khz. Nothing else is allow it seems.

Here is the startup log:

```
24.02.2018 19:13:53.965 (I) SDRangel 3.12.0 Qt 5.9.1 64b x86_64 Ubuntu 17.10 DSP Rx:16b Tx:16b PID 28502
24.02.2018 19:13:54.602 (W) qUncompress: Input data is corrupted
24.02.2018 19:13:54.602 (W) qUncompress: Input data is corrupted
24.02.2018 19:13:54.602 (W) qUncompress: Input data is corrupted
24.02.2018 19:13:54.602 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzer.so
24.02.2018 19:13:54.603 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzerng.so
24.02.2018 19:13:54.603 (I) PluginManager::loadPluginsDir: loaded plugin libdemodam.so
24.02.2018 19:13:54.604 (I) PluginManager::loadPluginsDir: loaded plugin libdemodatv.so
24.02.2018 19:13:54.604 (I) PluginManager::loadPluginsDir: loaded plugin libdemodbfm.so
24.02.2018 19:13:54.625 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddatv.so
24.02.2018 19:13:54.626 (I) PluginManager::loadPluginsDir: loaded plugin libdemodlora.so
24.02.2018 19:13:54.627 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnfm.so
24.02.2018 19:13:54.627 (I) PluginManager::loadPluginsDir: loaded plugin libdemodssb.so
24.02.2018 19:13:54.628 (I) PluginManager::loadPluginsDir: loaded plugin libdemodtcpsrc.so
24.02.2018 19:13:54.628 (I) PluginManager::loadPluginsDir: loaded plugin libdemodudpsrc.so
24.02.2018 19:13:54.629 (I) PluginManager::loadPluginsDir: loaded plugin libdemodwfm.so
24.02.2018 19:13:54.629 (I) PluginManager::loadPluginsDir: loaded plugin libmodam.so
24.02.2018 19:13:54.668 (I) PluginManager::loadPluginsDir: loaded plugin libmodatv.so
24.02.2018 19:13:54.669 (I) PluginManager::loadPluginsDir: loaded plugin libmodnfm.so
24.02.2018 19:13:54.669 (I) PluginManager::loadPluginsDir: loaded plugin libmodssb.so
24.02.2018 19:13:54.670 (I) PluginManager::loadPluginsDir: loaded plugin libmodudpsink.so
24.02.2018 19:13:54.670 (I) PluginManager::loadPluginsDir: loaded plugin libmodwfm.so
24.02.2018 19:13:54.671 (I) PluginManager::loadPluginsDir: loaded plugin liboutputfilesink.so
24.02.2018 19:13:54.672 (I) PluginManager::loadPluginsDir: loaded plugin liboutputlimesdr.so
24.02.2018 19:13:54.673 (I) PluginManager::loadPluginsDir: loaded plugin libinputfcdpro.so
24.02.2018 19:13:54.673 (I) PluginManager::loadPluginsDir: loaded plugin libinputfcdproplus.so
24.02.2018 19:13:54.674 (I) PluginManager::loadPluginsDir: loaded plugin libinputfilesource.so
24.02.2018 19:13:54.674 (I) PluginManager::loadPluginsDir: loaded plugin libinputlimesdr.so
24.02.2018 19:13:54.675 (I) PluginManager::loadPluginsDir: loaded plugin libinputrtlsdr.so
24.02.2018 19:13:54.676 (I) PluginManager::loadPluginsDir: loaded plugin libinputtestsource.so
Reference clock 40,00 MHz
SetPllFrequency: timeout, busy bit is still 1
Reference clock 40,00 MHz
SetPllFrequency: timeout, busy bit is still 1
24.02.2018 19:14:02.222 (I) WebAPIServer::start: starting web API server at http://127.0.0.1:8091
Reference clock 40,00 MHz
SetPllFrequency: timeout, busy bit is still 1
Rx calibration finished
Rx calibration finished
SetFrequencyCGEN(160 MHz) failed
24.02.2018 19:14:18.260 (C) LimeSDRInput::applySettings: could not set sample rate to 5000000 with oversampling of 8
LNAL has no connection to RF ports
SetFrequencyCGEN(320 MHz) failed
24.02.2018 19:14:18.528 (C) LimeSDRInput::applySettings: could not set frequency to 0
MCU error code(6): Invalid Rx path
RX LPF configured
24.02.2018 19:14:18.866 (W) AudioOutput::start: using default device alsa_output.pci-0000_00_1f.3.analog-stereo
24.02.2018 19:14:18.866 (I) AudioOutput::start: audio format OK
MCU error code(6): Invalid Rx path

```

f4exb, 2018-02-25:
Judging by the messages there is something quite wrong with the Mini. Maybe try to see if it works with the Lime Suite GUI first. In any case you should checkout the commit ebd027169defc9689d1eefa0f57d4df617b7cc4d of Lime Suite and not take the HEAD of master.

sgoudelis, 2018-02-25:
Ok so I tried the LimeSDR Suite GUI (see screenshot) and the FFT viewer does not work at all. I only saw the FFT viewer work once and not ever again. Could you please advice me if you can where to go from here? I guess this is not the place do offer support but I am looking for an opinion on the matter. 

Screenshot
https://imgur.com/a/WhrAC

f4exb, 2018-02-25:
I suggest you seek help in this forum: https://discourse.myriadrf.org/
The problem specific to the Mini is already tracked here: #133

martywittrock, 2018-02-25:
Please see the following video and picture links for the Lime-Mini playing perfect on SDRAngel hosted on a UDOO Advanced card using Ubuntu 17.10 (Artful) and with all the LimeSuite, gr-osmosdr, and SoapySDR dependencies installed and operational.

Video: https://youtu.be/m9fIFdxezF0

Picture: https://drive.google.com/file/d/1TK4iePpYJfMtfc8IBV47cFC4TM_Z6oh5/view?usp=sharing

Please check your installation of the following: LimeSuite, gr-osmosdr, and SoapySDR. Be sure to have a clear SoapySDRUtil --find    that shows the Lime-Mini in the list. Also ensure you have LimeSuite installed correctly by this link:

https://wiki.myriadrf.org/Lime_Suite

73 de Marty, KN0CK

f4exb, 2018-03-11:
The entire support with Lime Suite is crap now. Therefore "official" support of LimeSDR will be discontinued from the next release until Lime Suite issues are fixed. It will be still possible to compile from source but LimeSDR plugins will be removed from all binary distributions.

martywittrock, 2018-03-12:
Edouard,

When I saw this message over the weekend, I immediately contacted Andrew Back on this to have him work with you on what the issue is. SDRAngel is the ONLY application suitable for Amateur Radio and the LimeSDR...Without it, we have absolutely nothing to work with...The 'LimeSDR C#' application is not ready for prime time (too brittle to use) and SDRConsole V3.0 has produced nothing of substance for transmit when it was promised to be working by 'late summer (of '17),..The ONLY application that's been not only useful, but what we're ALL designing our LimeSDR rigs to IS SDRANGEL...I would hate to see the development stop in its tracks and not advance if we needed a unified Limesuite.dll that was controlled strictly by Lime Microsystems. Please let us know what the issue is (other than the support being 'crap') and I'll be happy to get priority on this from the Lime Team...In fact, Andrew was supposed to contact you over the past 24 hours - has that happened yet? Please advise and DON'T GIVE UP ON LIME..! 

73 de Marty, KN0CK

neeocf, 2018-03-12:
I've found the cause of this issue in LimeSuite API and reported it with proposed patch here: https://github.com/myriadrf/LimeSuite/issues/171

f4exb, 2018-03-13:
Thanks! I think this has been promptly fixed:
https://github.com/myriadrf/LimeSuite/commit/67d88c54690eecf6b73f0a06ab5e58429926bc6d

Then I can check for any other possible issues since this one was particularly annoying.

neeocf, 2018-03-13:
maybe some kind of assert if max freq > min would be usefull for the future?

f4exb, 2018-03-13:
I believe the Lime Suite library should take care of not returning nonsense.

f4exb, 2018-03-14:
The "big" lime appears to be working correctly now after inverting up/down conversion switch in the NCO control routine as the new version of the library actually inverted it.

unixpunk, 2018-03-15:
LimeSDR Mini is working as well, at least in initial testing.  Thanks!

martywittrock, 2018-03-15:
Edouard,

So how does this recent change in LimeSuite affect 'Angel..? Win and Linux, too? Would I need to recompile (Linux) and replace (LimeSuite.dll) for Windows? Are the changes implemented on Github just to compile LimeSuite again and then recompile 'Angel..? Let me know at your soonest - thanks,

73 de Marty, KN0CK

f4exb, 2018-03-15:
For binary releases the LimeSuite library is always included so there is no external dependency. If you compile from source you will have to take the latest version of the master branch of LimeSuite in Github compile and install it before compiling SDRangel. 

I have just succeeded in compiling a Mingw32 version of LimeSuite on Windows (needs a bit of tweaking) and the changes haven't reached this repository yet so for now consider this as Linux only. On Windows it still uses libusb so you will have to install Zadig drivers as it used to be. I still need to try if this works on Windows. 

Hopefully 3.13.0 will be out very soon with these latest changes (and more!). As far as I can see the Lime is doing very well with this new LimeSuite and I see that unixpunk has succeeded with the Lime Mini so that's quite encouraging.

In any case you will need to flash with the latest firmware.

f4exb, 2018-03-15:
The picture is not as pleasant on the Tx side. Modulation is bad. Absolutely no idea why other Tx devices like the HackRF work fine.

f4exb, 2018-03-16:
So the present status is that only single Rx works witl LimeSDR in SDRangel.

f4exb, 2018-03-17:
Leave the issue open as this still does not work satisfactorily

f4exb, 2018-03-20:
Using LMS_SetClockFreq() with LMS_CLOCK_SXR / LMS_CLOCK_SXT clock parameters in place of LMS_SetLOFrequency() does the trick. So channel A/B frequency issue will get fixed in the next version.

I will open a new issue for the Tx issue if it persists.

f4exb, 2018-03-24:
Rx frequency issues now fixed with v3.13.1

closed 2018-03-24 by f4exb

wgaylord, 2018-10-09:
Any suggestion for fixing this other then compiling LimeSDR Suite from source?

I have SDRAngle

> 2018-10-09 10:31:50.437 (D) using qt5ct plugin
> SDRangel 4.1.0

And I can't tune below 30 Mhz with my LimeSDR Mini

My LimeSuite version is 

> Version: 18.10.0-myriadrf1~bonic
> Build date: 2018-10-04

On Linux Mint 
Output from uname -a
> 4.15.0-20-generic #21-Ubuntu SMP Tue Apr 24 06:16:15 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

f4exb, 2018-10-12:
LimeSDR does not actually tune bleow 30 MHz. The trick to get a signal below the 30 MHz lime is to use the NCO. You have to tune the main LO to around 30 MHz then tune the NCO so that the center frequency of reception is moved downwards in frequency. Of course the RF filter has to be tuned accordingly so that its lower cutoff is not above the frequency you wish to receive.

wgaylord, 2018-10-12:
So they lied to everyone on the campaign....

f4exb, 2018-10-13:
The various amplifiers and mixers do work in the HF range it's only the main LO that will not go below 30 MHz. Considering the range (30 - 3800 MHz) this should not be a surprise. This is already a very large span for a single LO.