f4exb/sdrangel #140: Mismatch in 'struct' vs. 'class'

gvanem, 2018-02-27:
I'm on a mission to try to build SDRangel on Windows using MSVC. Realising that this is next to impossible now (variable length arrays and what not) , I've settled on the next-best thing; the wonderful `clang-cl` drop-in replacement. But even this returns several warning this these (with `-W3`):
```
plugin/pluginmanager.h(59,58):  warning: struct 'DeviceUISet' was previously declared as a class
      [-Wmismatched-tags]
    void createRxChannelInstance(int channelPluginIndex, struct DeviceUISet *deviceUISet, class DeviceSourceAPI ...
                                                         ^
plugin/plugininterface.h(19,7):  note: previous use is here
class DeviceUISet;
      ^
```
I think that due to this mismatch, the MSVC linker fails to link several .DLLs (mismatch in C++ mangled name). But it seems GCC is more liberal here; producing the same C++ mangling for both versions.

What is your intention? `class` or `struct`? I get this warning from approx. 20 places!

f4exb, 2018-02-27:
As it is forward declared as a class in many places let's make it a class in its own right then. I guess gcc makes no difference because in C++ there is not much difference but it never hurts to be consistent. Moving to clang-cl vs MSVC is probably a good move.

f4exb, 2018-02-27:
Fix is now on dev branch

gvanem, 2018-02-27:
Perhaps keep all those `class XX;` forwards in one common `.h`-file?

> Moving to clang-cl vs MSVC is probably a good move.

Great! The biggest hurdle with MSVC was in fact code like this:
```c
GLfloat q3[2*(end -start)];
```
Could it be done using `alloca()` or some STL-code instead?

gvanem, 2018-02-27:
So far, I've identified these warnings regarding `-Wmismatched-tags` in other places:  
```
../httpserver\staticfilecontroller.h(45,1):  warning: class 'HttpDocrootSettings' was previously declared as a struct [-Wmismatched-tags]
webapi/webapiserver.h(25,5):  warning: class 'HttpListenerSettings' was previously declared as a struct [-Wmismatched-tags]
../sdrbase\webapi/webapiserver.h(25,5):  warning: class 'HttpListenerSettings' was previously declared as a struct [-Wmismatched-tags]
../sdrbase\dsp/cwkeyersettings.h(24,1):  warning: 'CWKeyerSettings' defined as a struct here but previously declared as a class [-Wmismatched-tags]
./chanalyzerngplugin.h(24,1):  warning: struct 'DeviceUISet' was previously declared as a class [-Wmismatched-tags]
./amdemodgui.h(12,1):  warning: struct 'DeviceUISet' was previously declared as a class [-Wmismatched-tags]
./objects/../amdemodplugin.h(23,1):  warning: struct 'DeviceUISet' was previously declared as a class [-Wmismatched-tags]
./atvdemodsettings.h(24,1):  warning: struct 'Serializable' was previously declared as a class [-Wmismatched-tags]
./bfmplugin.h(24,1):  warning: struct 'DeviceUISet' was previously declared as a class [-Wmismatched-tags]
./datvdemodplugin.h(25,1):  warning: struct 'DeviceUISet' was previously declared as a class [-Wmismatched-tags]
```

f4exb, 2018-02-28:
> Perhaps keep all those class XX; forwards in one common .h-file?
I think this is ugly. The forward declaration should be where they are used for.

f4exb, 2018-02-28:
> GLfloat q3[2*(end -start)];

Firstly I was amazed that it worked but apparently this is a g++ thing that is not allowed in standard C++. As compilation with -pedantic reports:

`sdrgui/gui/glscope.cpp:330:34: warning: ISO C++ forbids variable length array ‘q3’ [-Wvla]
    GLfloat q3[4*tickList->count()];`

I am not sure `alloca()` is the right answer as its usage is a bit controversial but this is something to look into.

gvanem, 2018-02-28:
> allocate on the heap with new and delete[].

This will be much slower than `alloca()` though. Why is `alloca()` controversial?

f4exb, 2018-02-28:
You can find a lot of posts in the "netterature". Like:
  - https://stackoverflow.com/questions/1018853/why-is-the-use-of-alloca-not-considered-good-practice
  - https://stackoverflow.com/questions/39839865/fixed-size-array-vs-alloca-or-vlas

Top concerns are stack overflow and portability issues. For these concerns VLAs are no better than alloca. So I am criticizing the use of VLAs on the stack in fact. Note that the sizes involved are usually small and that the original authors (it is not my own code) have been careful not to allocate within loops and enclose the code sections in braces to control the scope. However in all cases the actual size of the arrrays depend on values of class members so the memory allocation/deallocation needs to take place only when these values change and that does not happen very often much less than where the arrays are used. Of course this needs code inspection and is more involving than just replacing VLA by alloca but this would improve code quality.

f4exb, 2018-03-01:
All VLAs removed in dev branch now.

gvanem, 2018-03-01:
Good. When will this be in `master`? I have trouble with a `git pull` from `dev`. But then I'm a git *n00b*.

f4exb, 2018-03-01:
This should work from your copy of sdrangel repository:
```git checkout dev```
```git pull origin dev```

gvanem, 2018-03-01:
Thanks a bunch. That worked. I'm trying to build `dev` now. Lot less warnings from clang-cl so far.
Mostly like these:
```
In file included from dsp/nullsink.cpp:1:
.\dsp/nullsink.h(16,15):  warning: 'NullSink::feed' hides overloaded virtual function [-Woverloaded-virtual]
        virtual void feed(SampleVector::const_iterator begin, SampleVector::const_iterator end, bool positiveOnly);
                     ^
...
In file included from mainwindow.cpp:63:
./webapi/webapiadaptergui.h(38,17):  warning: 'WebAPIAdapterGUI::instanceDelete' hides overloaded virtual function
      [-Woverloaded-virtual]
    virtual int instanceDelete(
                ^
```

f4exb, 2018-03-03:
Fix for `-Woverloaded-virtual` is on dev branch now

gvanem, 2018-03-03:
> Fix for -Woverloaded-virtual is on dev branch now

Good, thanks. What about the other warnings? Have you tried with clang/clang-cl or even MSVC? Then you'd also discover the DLL export/import issues.

f4exb, 2018-03-04:
Compiled with clang 7.0.0 on Linux with no warnings. I had to change the main CMakeLists.txt so that it would detect clang correctly. The compilation should fail on cm256cc includes without this change. Tried the binary and it works (as one would expect) and is very slightly better in terms of CPU load than the gcc version. However the gcc used is 5.3.0 so a fairly old version.

So it is now on dev branch.

gvanem, 2018-03-04:
Just curious; does `clang-cl` exist on Linux too? Maybe that could detect more issues? Or if you somehow disable the [auto-export](https://access.redhat.com/documentation/en-US/Red_Hat_Enterprise_Linux/4/html/Using_ld_the_GNU_Linker/win32.html) of classes feature when cross-compiling MinGW on Linux?

f4exb, 2018-03-04:
I tried `clang-cl` on Linux (it exists) but using it just in place of `clang++` does not work because cmake fails early being unable to create an executable. With a proper cross compilation environment this may work.

f4exb, 2018-03-17:
Done in v3.13.0

closed 2018-03-17 by f4exb

srcejon, 2025-07-07:
> Sorry to comment on an old issue. But there are still issues with this. E.g. in `plugins/channelrx/demoddatv/datvdemodgui.h`. Building with `cl` or `clang-cl`, game this link error:

Not a problem with latest version of cl for me, but have submitted a patch.