f4exb/sdrangel #141: DLL exports for MSVC/clang-cl

gvanem, 2018-03-01:
I know SDRangel is currently built using MinGW on Windows. This should be okay with the *automagical* (but IMHO ugly and confusing) [auto-export](https://access.redhat.com/documentation/en-US/Red_Hat_Enterprise_Linux/4/html/Using_ld_the_GNU_Linker/win32.html) feature of GNU's linker.

Trying to build e.g. `sdrgui.dll` using MSVC/clang-cl, leads to all kinds of troubles since there is only one `sdrbase/util/export.h` that sets `SDRANGEL_API` on the classes.

I suggest one does as in GnuRadio with an `api.h` header in every `include/gnuradio/*` directory that controls the `dllexport` and/or `dllimport`. Here is the easiest fix I can image (a `sdrbase/sdrbase.h`):
```c
/* Only 1 of this; defines what
 * __SDR_EXPORT and __SDR_IMPORT should be for all supported compilers
 */
#include <util/export.h> 
#ifdef sdrbase_EXPORTS   /* Set when building sdrbase.dll */
  #define SDRBASE_API __SDR_EXPORT
#else   /*  When using sdrbase.dll (i.e. linking to sdrbase.lib) from the outside */
  #define SDRBASE_API __SDR_IMPORT
#endif
```

Or add the above to the *chief header* in each directory, but I fail to see which one those are. Maybe a layout like this instead:
* `<root>/sdrbase/sdrbase.h` including `sdrbase/util/export.h`
* `<root>/sdrgui/sdrgui.h` including `sdrbase/util/export.h`
* etc.

f4exb, 2018-03-02:
Thatś the problem when you try to get MSVC in the picture and one of the reasons I hate it. It makes you add tons of defines and stuff that uglifies the code. I am not too keen on adding this stuff but I will have a closer look.

gvanem, 2018-03-02:
> Edit: are you sure it is a problem with sdrgui?

Yes. And swagger classes doesn't AFAICS have any dllexport/dllimport decoration on them. 
Besides it is generated code. Therefore I'm using swagger as a static lib for now.

I've come up with a`sdrbase/util/export.h` like [this](https://gist.github.com/gvanem/1d313ded7cb8bb58ef9fb060d7223085).

And have added the needed `x_API` to many .h-files. Now all `sdrangel.exe` and the plugins links fine using clang-cl and the MSVC linker.

f4exb, 2018-03-03:
Having all defines centralized in `sdrbase/util/export.h` is acceptable. I have modified it pretty much the way you did with a few cosmetic changes. I also removed `SDRANGEL_API` as it is now obsolete.

I am not sure all necessary changes have been made since I have no way to check at present. How do you build with `clang-cl`? Do you use Qt Creator with a custom toolchain?

Anyway changes are now committed on dev branch.

gvanem, 2018-03-04:
> Do you use Qt Creator with a custom toolchain?

No, the `.pro` files are not ready for MSVC/clang-cl. They all contains only `MINW32` and `MINW64`. 
I use a set of hand-make GNU-makefiles.

f4exb, 2018-03-17:
Done in v3.13.0

closed 2018-03-17 by f4exb

gvanem, 2018-03-23:
Hi again. There is still missing exports of several classes.