f4exb/sdrangel #142: Compiling inputPerseus.dll plugin

gvanem, 2018-03-05:
I've been able to compile and link almost all plugins. But hit the following compile error I fail to understand:
```
In file included from PerseusThread.cpp:19:
In file included from ./perseusthread.h:26:
../../../sdrbase\dsp/decimators.h(1096,16):  error: no matching member function for call to 'myDecimate'
                m_decimator2.myDecimate(
                ~~~~~~~~~~~~~^~~~~~~~~~
PerseusThread.cpp(102,24):  note: in instantiation of member function 'Decimators<long long, TripleByteLE<qint64>, 16,
      24>::decimate2_cen' requested here
        m_decimators64.decimate2_cen(&it, (TripleByteLE<qint64>*) buf, len);
                       ^
../../../sdrbase\dsp/inthalfbandfiltereo1.h(566,10):  note: candidate function not viable: no known conversion from
      'long long *' to 'int32_t *' (aka 'int *') for 3rd argument
    void myDecimate(int32_t x1, int32_t y1, int32_t *x2, int32_t *y2)
         ^
../../../sdrbase\dsp/inthalfbandfiltereo1.h(556,10):  note: candidate function not viable: requires 2 
arguments, but 4 were provided
    void myDecimate(const Sample* sample1, Sample* sample2)
         ^
```
And:
```
In file included from PerseusThread.cpp:19:
In file included from ./perseusthread.h:26:
../../../sdrbase\dsp/decimators.h(1154,16):  error: no matching member function for call to 'myDecimate'
                m_decimator4.myDecimate(
                ~~~~~~~~~~~~~^~~~~~~~~~
../../../sdrbase\dsp/inthalfbandfiltereo1.h(566,10):  note: candidate function not viable: no known conversion from
      'long long *' to 'int32_t *' (aka 'int *') for 3rd argument
    void myDecimate(int32_t x1, int32_t y1, int32_t *x2, int32_t *y2)
         ^
../../../sdrbase\dsp/inthalfbandfiltereo1.h(556,10):  note: candidate function not viable: requires 2
 arguments, but 4 were provided
    void myDecimate(const Sample* sample1, Sample* sample2)
         ^
```
My lack of C++ prevents me from fixing this myself 🤒.

But it seems some template function in `dsp/decimators.h` is missing.

*BTW*. Correct me here, but AFAICS a template-function in e.g. a plugin .cpp-file is only instantiated when called.  Thus there *must* be no `SDRBASE_API` on template-definitions in `dsp/*.h` . (these should be considered header-only code).  I guess this means a plugin.dll becomes bigger than needed. Maybe all instance-combinations could be defined in `sdrbase.dll` and exported from there?

f4exb, 2018-03-05:
You must not have the `SDBASE_API` decorator on a template if it has no concrete implementation. This is not the case for some of them like `HBFIRFilterTraits<16>` that have some code defined in the accompanying .cpp file. But any such problem would appear at link time and the error you get is at compile time. As it runs smoothly on Linux both with gcc and clang I suppose there is some obscure definition somewhere hiding parts of code to clang-cl. 

The particularity with Perseus is that it uses the `TripleByteLE` template which has only concrete implementations defined in the header `dsp/decimators.h` There may be a decorator missing here but again I expect any issue to show up at link time only.

gvanem, 2018-03-06:
> I suppose there is some obscure definition somewhere hiding parts of code to clang-cl.

Seems the obscurity is in `<qglobal.h>`:
```
typedef __int64 qint64;            /* 64 bit signed */
```
versus `long long` elswhere. Including for GCC on Windows.

But building with `-DSDR_RX_SAMPLE_24BIT` works. I fail to understand why.
Doesn't 24-processing decrease performance on e.g. rtlsdr and SDRPlay?

f4exb, 2018-03-06:
> Doesn't 24-processing decrease performance on e.g. rtlsdr and SDRPlay?
In fact that's the reason why it is not the default until I have put it in more scrutiny. I am not worried about relatively slow devices like RTLSDR and SDRplay. In fact it is geared towards the "slow" (under 2 MS/s) AirspyHF and Pereseus. These do need samples larger that 16 bits to best perform or just perform correctly (Perseus). In fact at these sample rates there is a slight increase in CPU load but not more than a few % scaled to one CPU so there is nothing to worry about. It is slightly more at 10 MS/s with the HackRF but I fail to run it at higher rates for a different reason. Ideally this should be a choice when you instantiate a new device set but this is a fairly large development and it should be worth more than a few CPU load % so we'll see. There is also the option to convert to float early that has also to be challenged.

Back to the original issue. Are you sure you have `-DSDR_RX_SAMPLE_24BIT` defined everywhere. It must be a global option. You cannot just have part of the code with it because the samples flow through all the application. Typically this would produce the kind of error you encounter.

gvanem, 2018-03-06:
> Are you sure you have -DSDR_RX_SAMPLE_24BIT defined everywhere? 

Yes, I have a `CFLAGS += -DSDR_RX_SAMPLE_24BIT` in my top-level hand-made `Common.Windows` makefile.

And BTW, now `SDRangel.exe` runs fine. Although some crashes when I hit the wrong button. Another issue coming up.

f4exb, 2018-03-18:
I suggest to drop Perseus supoort for clang-cl / Windows. Hence closing this issue.

closed 2018-03-18 by f4exb