f4exb/sdrangel #143: Scan feature

brendanthebig, 2018-03-06:
Is it possible to add a scan feature to some demod plugins. Possibly based on activity on preset frequencies or signal above certain level in passband ? Would be a very handy feature.

f4exb, 2018-03-06:
This is a very complex implementation that is not in line with the overall design. You can have as many channels as you want so there is no real need for this feature.

f4exb, 2018-03-08:
What you have to understand is that SDRangel is not monolithic in the sense that the demodulator and RF front end processing are all integrated in one single module. It makes it more flexible as you can add as many demodulators even of different kind in the passband obtained from the RF module. This also makes it more difficult to control the RF module from any of the channel demodulators and firstly which one? If implemented "simply" as you suggest  I can only see this ending in an ugly mishmash.

And firstly why would you need to scan "frequencies" one by one when you can process them all in parallel? I think you are too much biased with old analog radios that by construction had to scan every frequency with their single (or double at best) "VFO" chain.

In any case there is an API with which you are (or will be when the full monty is in place) able to control the frequency of the RF part and the shift of every channel. The only part missing is squelch reporting but this is in the plans. So I would leave it to external programming really.

The need for frequency sweeping and modules collaboration could be motivated by adding network analysis capabilities which I keep in a corner of my mind but I don't see "scanning" as the prime driver.

brendanthebig, 2018-03-08:
The only problem I have with adding multiple demodulators is that the audio from each demodulator exits via the same audio sink device. This can be a problem if piping the audio to an external program for further processing and multiple demodulators are simultaneously active. Not a big problem but I can imagine it would be a major issue to get each demodulator to have its own audio sink device. As far as scanning goes, I can appreciate the complexity involved due to the way sdrangel works. I agree that we can implement our own scanning features with the API and am looking forward to playing around with it when "the full monty is in place"..

Keep up the good work Edouard.

f4exb, 2018-03-09:
I imagine it would be possible to associate the audio output by demodulator instead of globally. This also represents a fairly big amount of work considering how this part is designed now. This could be the theme of another ticket.

brendanthebig, 2018-03-12:
Thanks for creating the new ticket re audio output by demodulator...I am closing this ticket now and am looking forward to experimenting with the API when the "full monty" version is released.

Once again, thanks for all of your great work on SDRAngel Edouard...

f4exb, 2018-03-19:
I am actually working on implementing the "report" API for channels limited to NFM modulator and demodulator for the moment. I am also writing a Python script to demonstrate a simple scanner control program connected to SDRangel via the REST API.

f4exb, 2018-03-24:
Implemented in v3.13.1 as an example of client script: `swagger\sdrangel\examples\scanner.py` This is fully functional for the supported plugins i.e devices: AirspyHF, HackRF, LimeSDR, RTLSDR and channels: AM demod, NFM demod.

closed 2018-03-24 by f4exb