f4exb/sdrangel #144: MSVC "fcdhid" fix

gvanem, 2018-03-06:
Currently `./fcdhid/hid-libusb.c` is barely Windows compatible and rather MinGW centric.
E.g. this doesn't parse with MSVC or clang-cl:
```c
hid_device * HID_API_EXPORT hid_open_path(const char *path)
```
(see below).

And the function `libusb_fill_interrupt_transfer()`mandates that on Windows, the `callback` *must* be `WINAPI` (== `__stdcall` == `LIBUSB_CALL`).

These patches makes it compile with MinGW, MSVC and clang-cl:
```diff
--- a/fcdhid/hid-libusb.c 2017-07-05 16:48:42
+++ b/fcdhid/hid-libusb.c 2018-03-06 18:36:02
@@ -37,7 +37,8 @@
 #include <unistd.h>
 #include <sys/types.h>
 #include <sys/stat.h>
-#ifndef MINGW32
+#include <sys/time.h>
+#ifndef _WIN32
 #include <sys/ioctl.h>
 #include <sys/utsname.h>
 #endif
@@ -65,6 +66,10 @@
 #define DETACH_KERNEL_DRIVER
 #endif

+#ifndef LIBUSB_CALL
+#define LIBUSB_CALL /* In case using some old libusb.h */
+#endif
+
 /**
  * MacOS does not implement POSIX Thread Barriers
  */
@@ -653,7 +658,7 @@
        return handle;
 }

-static void read_callback(struct libusb_transfer *transfer)
+static void LIBUSB_CALL read_callback(struct libusb_transfer *transfer)
 {
        hid_device *dev = transfer->user_data;
        int res;
@@ -791,7 +796,7 @@
 }

-hid_device * HID_API_EXPORT hid_open_path(const char *path)
+hid_device HID_API_EXPORT *hid_open_path(const char *path)
 {
        hid_device *dev = NULL;
```

f4exb, 2018-03-06:
You can forget about MinGW. This part is used for the FCD which is not supported.

gvanem, 2018-03-07:
> ... when trying to interface with ALSA which is Linux only.

In the `inputFcDpro.dll` plugin that uses *fcdhid*, yes I saw that ALSA stuff. Does a FunCube really need ALSA as a *input sampling* library? 

I took a look at some old SDR-Sharp sources. That uses `"*DirectSound.*fcd"` as hint for a FCD-device and let PortAudio open that using ` Pa_OpenStream()`. And there is e.g. `gr-fcd` which uses `gr-audio` as an input library.  I believe GnuRadio also uses PortAudio. 

The issue is not so important for me as I do not have a FunCube. But a SDRPlay. I would be nice if I could get that working with SDRangel (the MiriSDR-4 lib does not work with that now).

f4exb, 2018-03-07:
SDRPlay is also Linux only for a reason I don't remember exactly. I think I could compile the plugin and even start it but it would not work properly.

f4exb, 2018-03-18:
I suggest to drop FCD support for clang-cl / Windows. Hence closing this issue.

closed 2018-03-18 by f4exb