f4exb/sdrangel #145: Audio output by demodulator

f4exb, 2018-03-09:
This issue is opened to handle the setting of audio output by demodulator individually. The UDP and RTP output should also be handled this way as a particular case of audio output.

texasyojimbo, 2018-03-20:
Hypothetically, would it be possible to "stack" two demodulators on the same frequency with different modes? For example an NFM demodulator and a DSD demodulator on the same freq of a Yaesu System Fusion repeater output?

Also can the same output be assigned to different demodulators, or would each one have to be unique? For example, in the case of a SystemFusion repeater, can both analog (from NFM demod) and digital (from DSD demod) audio go to the same soundcard/speaker/UDP port?

P.S. Not sure DSD supports Yaesu System Fusion because of Yaesu weirdness, I am just throwing this out as a possible use-case of stacked demodulators.

f4exb, 2018-03-20:
Your first question is unrelated to the subject: yes you can stack two demodulators on the same frequency and that's independent of any audio consideration. Each demod being independent with its own frequency shift you can put them anywhere including stacking them on the same frequency.

At present the audio output of all demodulators are mixed on the same audio sink (the only one that you select in the audio options) so the scenario you are planning works. The purpose of this ticket is to give the ability to route the audio of a demodulator to a particular audio sink where it will be possibly mixed with the output of demodulators using the same sink. 

For DSD and YSF it works except for the wide mode that indeed introduces some weirdness that even defeats the genuine AMBE hardware decoder.

f4exb, 2018-03-20:
Note that with pulseaudio it is possible to create a null sink and associate it to a RTP sink also provided by pulseaudio. This makes a nice alternative to the present RTP sink that works only marginally (no FM stereo, no DSD). This would also allow to RTP send an audio mix of channels not just a single channel.

f4exb, 2018-03-23:
Preliminary works:
  - move AudioDeviceInfo from MainCore and MainWindow to DSPEngine
  - make AudioDeviceInfo manage AudioOutput(s) and AudioInput(s) and their AudioFifo allocation
  - maybe rename AudioDeviceInfo to AudioDeviceManager

f4exb, 2018-03-26:
Will also include sample rate selection and copy to UDP/RTP will be moved there.

f4exb, 2018-03-30:
implemented in v3.14.0

closed 2018-03-30 by f4exb