f4exb/sdrangel #149: FreeDV / Codec2

oh3gai, 2018-03-19:
Hi! 

First of all thanks for the awesome application. I was easily able to install it for both ubuntu and win7.
Is it possible to integrate FreeDV/Codec2 to demodulators and sinks? I think this would have huge impact for popularity of amateur digital voice modes.

Great work!

Br, Jarno

DL4APT, 2018-03-19:
FreeDV is a grate thing. If i have some time, I will test if sound piping from an SSB demod via UDP is a first step solution to get it running. A "bad" but working solution could be virtual audio cable. Native suport still would be best but on can only work at some thing at the same time.

herrfeuer, 2019-02-10:
It would be great if Codec2 or Opus could be used instead of AMBE for DMR or C4FM instead of mbelib.

f4exb, 2019-02-20:
Opus is now in the dependencies but used to send the audio stream through the network. Opus has a minimal sample rate of 6 kbits/s. For best compatibility one should aim at 72 bits per 20 ms frame which sets the limit to 3.6 kbit/s. This is slightly above the upper limit of 3.2 kbit/s of Codec2 so one could imagine adding some FEC or slowing down the rate to improve robustness. Codec2 works still very well at 1.6 kbits/s. Maybe D-Star and GMSK would be the first thing to try. At half rate one could aim at a ~3kHz bandwidth. A quarter rate is even possible since it still works at 0.8 kbits/s. In order to conduct these experiments transmission support is necessary because of the lack of commercial equipment. A ticket has been open in the DSDcc project to this purpose.

The purpose of this one is FreeDV though.

DL4APT, 2019-03-02:
Thanks for the addition will test it in the coming days

f4exb, 2019-03-02:
Implemented in v4.5.0. A few things still need to be ironed out but will come later.

closed 2019-03-02 by f4exb