f4exb/sdrangel #150: NGscope tracking (feature)

F5OEO, 2018-03-20:
Could be usefull to allow DATV process to drive the triger NGscope : frequency/symbolrate could allows to view a real eye diagram. Could be extensible to other digital modulation.

f4exb, 2018-03-21:
Had the idea for a long time. Now the DATV demodulator and LeanSDR could bring more concrete re-usable examples.

f4exb, 2018-04-02:
Actually [liquid-dsp](https://github.com/jgaeddert/liquid-dsp) could be more suitable for this purpose.

f4exb, 2018-04-02:
Nor liquid-dsp nor leansdr seem to fit
  - Leansdr lacks documentation and seems too far from the scope logic
  - Liquid-dsp is unfit to be integrated in the project (no cmake, no packaging) 

So all functions have to be re-implemented internally. This is not going to happen anytime soon.

f4exb, 2018-04-04:
After some trials it appears that liquid-dsp does not work on concrete cases (Tetra signals) and leansdr is not adapted for clock recovery. Hence the algorithms have to be coded from theory and this is going to happen anytime soon.

f4exb, 2018-05-23:
Since v3.14.7 NG scope has carrier recovery option for PSK signals from 1 (CW) to 16 equally spaced phases in powers of two (1,2,4,8,16). There is no symbol clock recovery however so this issue would call for symbol clock recovery more specifically.

f4exb, 2018-05-25:
Thanks for the illustrative screenshot. As you can see phase 0 is always represented unlike the figure of the DATV demodulator which represents more conventionally the points rotated by pi/4. This is due to the phase comparator of the PLL that always has 0 as one of its points. As it is only a matter of convention I made the choice of simplicity: multiply the phase error by the modulation order (1,2,4,8,16) and take modulo 2*pi. This is a brilliant idea that I have taken from this post: https://www.dsprelated.com/showthread/comp.dsp/36356-1.php and easily converts a standard software PLL to a multi-PSK PLL.

closed 2019-02-20 by f4exb