f4exb/sdrangel #151: New feature request: support UHD for Ettus USRP radios

courtarro, 2018-03-21:
These radios are quite common at my office, so it would be great to play with them inside sdrangel. Full support exists within GnuRadio in the form of the `gr-uhd` module, and within the `osmosdr` driver.

As an alternative, I wonder if it's possible to build generic `osmosdr` source/sink plugins to support any radios available by that driver.

f4exb, 2018-03-22:
If you can send me one of these (expensive) radios I may look into it :smile: 

More seriously I already have in the back of my mind the idea to support the osmo/soapy/whatever family of "generic" drivers. There is nothing like a "generic" driver for SDR hardware in my opinion so for me these are only approximations which explain why I have been reluctant or at best not showing much interest in supporting these.

I only looked a little bit more into Soapy and this calls for a different solution than the usual plugins. AFAIK Soapy combines streams of the same hardware device where SDRangel sees one stream as one device. What would be possible is to create some "virtual" devices at top level that would represent one stream of the instantiation of a real device with Soapy. Then this "virtual" device could be presented in the list of devices along with the "real" ones and used by a generic Soapy device plugin.

Then it would be possible to test the Soapy thing with some hardware I own that is supported by Soapy and hope for the best for other devices like UHD.

This represents a fairly large amount of work and cannot be planned any time soon.

longyucheng123, 2018-03-25:
Hello! Can you add audio resources to the sdrangel software for launch? (for example~.wav or~.mp3 file resources), thank you!

f4exb, 2018-03-26:
In short: no

ghulands, 2018-05-19:
https://github.com/pothosware/SoapySDR supports the USRP along with a whole host of other devices. It might be worth having Soapy as the primary driver so you can focus on features in SDRAngel.

f4exb, 2018-09-28:
Support (not guaranteed) will be done via SoapySDR.
Closed in favor of #221

closed 2018-09-28 by f4exb