f4exb/sdrangel #155: Feature suggestion: Frequency translation for limesdr

texasyojimbo, 2018-04-07:
I note that some of the other sources/sinks have a transverter mode. For example RTLSDR.

One can use transverters with the LimeSDR; one can also use transmitted harmonics (for example 3rd harmonic of 3456 is 10368) with filtering to transmit very low power signals on higher bands.

Would be nice to have a way to adjust the dial frequency by either adding/subtracting the LO or by multiplying.

Low priority "nice to have."

f4exb, 2018-04-08:
Note that if you use harmonics this is not the same as using a transverter in front of the Lime. The transverter mode implemented for other device is a linear translation not a multiplication.

texasyojimbo, 2018-04-08:
Duly noted and agreed.

I just see from a user standpoint two different ways in which the displayed
frequency could be offset/changed to improve the user experience
(transverter style addition and subtraction, as well as multiplication
using a frequency multiplier or tx harmonic).

I am using a mixer to move LF/MW signals up to VHF where the LimeSDR is
more sensitive; and am using 3rd harmonics (with filtering of the
fundamental) to generate microwave signals. In theory I could do both; for
example using the 3rd harmonic of 3456 MHz as an LO for a mixer to receive
10Ghz.

texasyojimbo, 2018-04-08:
Or to clarify I am describing two different but not exclusive use cases. :)

f4exb, 2018-04-09:
I suggest to drop the harmonic operation case. It has side effects that lead to excessively complex processing when not presenting completely wrong data. Basically this would work fine if you are in FM mode at the center frequency but would be wrong in almost every other case. Even though you would have to consider spectrum expansion to be accurate. So I'd rather leave the user with the maths of harmonic operation and limit UI support to linear translation (mixing) which is also simple to implement as it has already been done for other plugins.

f4exb, 2018-04-20:
Implemented in v3.14.3

closed 2018-04-20 by f4exb