f4exb/sdrangel #157: Crash when parsing RDS

jigpu, 2018-04-10:
Some local FM stations seem to cause sdrangel to crash shortly after enabling RDS decoding. I can't understand what's going on, but some of the code definitely looks suspicious (e.g. the "while" loop might result in ff_pointer going negative? doesn't look like that's the problem though).

To make reproducing easier, I captured the IQ data from the station with RDS disabled and then truncated the file to be only a little longer than necessary to trigger the crash with RDS enabled.

Tested with revision

 306c09e1757312dbdfe44f3acd3aba70acbf17bf

Reproduction:
1. Unzip the attached IQ capture
2. Open sdrangel
3. Add a Broadcast FM Demodulator
4. Enable stereo decoding
5. Enable RDS decoding
6. Load the unzipped IQ capture
7. Hit play
8. After ~1.5 seconds, sdrangel crashes

Backtrace & some additional locals:
~~~
Thread 14 "QThread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff9a8c9700 (LWP 21699)]
0x00007fffd5151889 in RDSParser::decode_optional_content (this=0x555556191e98, no_groups=<optimized out>, free_format=0x7fffd5386da0 <RDSParser::decode_type8(unsigned int*, bool)::free_format>)
    at /home/jgerecke/.cache/pacaur/sdrangel-git/src/sdrangel/plugins/channelrx/demodbfm/rdsparser.cpp:942
942				content_length = optional_content_lengths[m_g8_label_index];
(gdb) bt full
#0  0x00007fffd5151889 in RDSParser::decode_optional_content(int, unsigned long*) (this=0x555556191e98, no_groups=<optimized out>, free_format=0x7fffd5386da0 <RDSParser::decode_type8(unsigned int*, bool)::free_format>) at /home/jgerecke/.cache/pacaur/sdrangel-git/src/sdrangel/plugins/channelrx/demodbfm/rdsparser.cpp:942
        i = <optimized out>
        content_length = <optimized out>
        ff_pointer = 17
#1  0x00007fffd5135906 in BFMDemod::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, bool) (this=0x555556191a20, begin=..., end={m_real = -3253, m_imag = 1383}, firstOfBurst=<optimized out>)
    at /home/jgerecke/.cache/pacaur/sdrangel-git/src/sdrangel/plugins/channelrx/demodbfm/bfmdemod.cpp:181
        bit = false
        r = <optimized out>
        sampleStereo = <optimized out>
        e = <optimized out>
        i = <optimized out>
        c = {_M_value = 0.119445801 + -0.0104980469 * I}
        it = {m_real = 3914, m_imag = -344}
        ci = {_M_value = 0 + 0 * I}
        cs = {_M_value = 0 + 0 * I}
        cr = {_M_value = 0.00799068343 + 0 * I}
        rf = 0x555556157da0
        msq = <optimized out>
        demod = -0.0109286103
#2  0x00007ffff7573765 in DownChannelizer::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, bool) (this=0x555556165300, begin={m_real = -3121, m_imag = 885}, end={m_real = 3478, m_imag = 1733}, positiveOnly=<optimized out>)
    at /home/jgerecke/.cache/pacaur/sdrangel-git/src/sdrangel/sdrbase/dsp/downchannelizer.cpp:88
#3  0x00007ffff75d0d7c in ThreadedBasebandSampleSinkFifo::handleFifoData() (this=0x5555565be0c0) at /home/jgerecke/.cache/pacaur/sdrangel-git/src/sdrangel/sdrbase/dsp/threadedbasebandsamplesink.cpp:45
        part2begin = {m_real = 0, m_imag = 0}
        part2end = {m_real = 0, m_imag = 0}
        count = <optimized out>
        part1begin = {m_real = -3121, m_imag = 885}
        part1end = {m_real = 3478, m_imag = 1733}
#4  0x00007ffff60c3062 in QObject::event(QEvent*) () at /usr/lib/libQt5Core.so.5
#5  0x00007ffff6e0ffec in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#6  0x00007ffff6e179c6 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#7  0x00007ffff6091cf0 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#8  0x00007ffff6094956 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib/libQt5Core.so.5
#9  0x00007ffff60eec54 in  () at /usr/lib/libQt5Core.so.5
#10 0x00007ffff1b08458 in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#11 0x00007ffff1b086a1 in  () at /usr/lib/libglib-2.0.so.0
#12 0x00007ffff1b0872e in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#13 0x00007ffff60ee241 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#14 0x00007ffff609032b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#15 0x00007ffff5e9f72e in QThread::exec() () at /usr/lib/libQt5Core.so.5
#16 0x00007ffff5ea4acd in  () at /usr/lib/libQt5Core.so.5
#17 0x00007ffff38d308c in start_thread () at /usr/lib/libpthread.so.0
#18 0x00007ffff5595e7f in clone () at /usr/lib/libc.so.6
(gdb) print m_g8_label_index
$1 = 1310720
(gdb) print *free_format
$2 = 1310720
(gdb) print ff_pointer
$3 = 17
~~~

[test_0-trunc.zip](https://github.com/f4exb/sdrangel/files/1895873/test_0-trunc.zip)

f4exb, 2018-04-11:
Thanks for the detailed report. RDS is a bit of a patchwork code with parts imported from other applications and it definitely needs some inspection. `ff_pointer = 17` is indeed very low and there must be some pointer calculation mishmash.

f4exb, 2018-04-11:
Problem identified. Was introduced with commit 324aaa8 and the change of `&&` by `&`. What was actually meant is to take the integer cast of a boolean which is better if made explicit (clang issues a warning and the purpose of the commit was to fix clang warnings). There is also some code around to fix PVS analyzer warning. It turns out that it can be simplified not to confuse the analyzer.

jigpu, 2018-04-12:
Thanks for the quick fix! I'm away from my radio for the next week but I'll definitely test it as soon as I can to see if there are other crashers still lurking on that station. 

BTW, based on your description, did you mean 723df69 instead of 324aaa8? Or did that commit make and && to & change break?

f4exb, 2018-04-12:
Yes you're right in fact 723df69 introduced the bug by changing `&&` to `&`.

f4exb, 2018-04-15:
Fixed in v3.14.2

closed 2018-04-15 by f4exb