f4exb/sdrangel #158: Storing device selection in presets.

brendanthebig, 2018-04-16:
Is it possible to store chosen device (sink or source) as well as chosen channel of device on multi-channel devices in presets ?

f4exb, 2018-04-17:
This request is problematic because it conflicts with the present functionality. The presets stores settings for the current device selected at save time in a list of device settings so when you use the preset with device A you retrieve the settings made for device A and when you use device B you retrieve those of device B. Thus allows using several devices with the same preset which I think is a valuable feature. 

So I suggest that instead you load the preset in two steps: select device first then load the preset. 

If you have a routine configuration you could create a Python script using the REST API. See https://github.com/f4exb/sdrangel/blob/master/swagger/sdrangel/examples/devicesets_config.py for an example. This is the one I am actually using and is of course just an example and you could write your own along the same lines.

f4exb, 2018-04-26:
Closing this as the current functionality should be kept with several devices able to share the same preset.

closed 2018-04-26 by f4exb