f4exb/sdrangel #159: Occasional crash when removing Broadcast FM Demodulator

jigpu, 2018-04-18:
sdrangel occasionally crashes if a Broadcast FM Demodulator is removed. This doesn't always happen -- only when closed at the "wrong" moment. I've seen a couple of different backtraces, but the one at the bottom of this report is by far the most common / reproducible.

Tested with revision 01798c9d

Reproduction:
1) Unzip the IQ capture from bug #157 
2) Open sdrangel
3) Load the unzipped IQ capture
4) Hit play
5) Add a Broadcast FM Demodulator
6) Close the demodulator
7) If the program is still running, go to step 5
8) If you're here, the program crashed :(

Backtrace & GDB data:
~~~
Thread 14 "QThread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff99491700 (LWP 17313)]
0x00007ffff75c5775 in fftfilt::runFilt (this=0x555556664e60, in=..., out=out@entry=0x7fff994906f8) at /home/jgerecke/.cache/pacaur/sdrangel-git/src/sdrangel/sdrbase/dsp/fftfilt.cxx:236
236		data[inptr++] = in;
(gdb) bt full
#0  0x00007ffff75c5775 in fftfilt::runFilt(std::complex<float> const&, std::complex<float>**) (this=0x555556664e60, in=..., out=out@entry=0x7fff994906f8)
    at /home/jgerecke/.cache/pacaur/sdrangel-git/src/sdrangel/sdrbase/dsp/fftfilt.cxx:236
#1  0x00007fffd5087cd3 in BFMDemod::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, bool) (this=0x555556655800, begin=..., end={m_real = 2520, m_imag = 2275}, firstOfBurst=<optimized out>)
    at /home/jgerecke/.cache/pacaur/sdrangel-git/src/sdrangel/plugins/channelrx/demodbfm/bfmdemod.cpp:130
        c = {_M_value = 0.00833129883 + -0.117736816 * I}
        it = {m_real = 273, m_imag = -3858}
        ci = {_M_value = 0 + 0 * I}
        cs = {_M_value = 0 + 0 * I}
        cr = {_M_value = 0 + 0 * I}
        rf = 0x555557304330
        msq = <optimized out>
        demod = 0.0914670229
#2  0x00007ffff7574595 in DownChannelizer::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, bool) (this=0x7fffe0016940, begin={m_real = -2851, m_imag = 2832}, end={m_real = -950, m_imag = 2482}, positiveOnly=<optimized out>)
    at /home/jgerecke/.cache/pacaur/sdrangel-git/src/sdrangel/sdrbase/dsp/downchannelizer.cpp:88
#3  0x00007ffff75d1dbc in ThreadedBasebandSampleSinkFifo::handleFifoData() (this=0x555556764220) at /home/jgerecke/.cache/pacaur/sdrangel-git/src/sdrangel/sdrbase/dsp/threadedbasebandsamplesink.cpp:45
        part2begin = {m_real = 0, m_imag = 0}
        part2end = {m_real = 0, m_imag = 0}
        count = <optimized out>
        part1begin = {m_real = -2851, m_imag = 2832}
        part1end = {m_real = -950, m_imag = 2482}
#4  0x00007ffff60c3062 in QObject::event(QEvent*) () at /usr/lib/libQt5Core.so.5
#5  0x00007ffff6e0ffec in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#6  0x00007ffff6e179c6 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#7  0x00007ffff6091cf0 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#8  0x00007ffff6094956 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib/libQt5Core.so.5
#9  0x00007ffff60eec54 in  () at /usr/lib/libQt5Core.so.5
#10 0x00007ffff1ae6458 in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#11 0x00007ffff1ae66a1 in  () at /usr/lib/libglib-2.0.so.0
#12 0x00007ffff1ae672e in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#13 0x00007ffff60ee241 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#14 0x00007ffff609032b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#15 0x00007ffff5e9f72e in QThread::exec() () at /usr/lib/libQt5Core.so.5
#16 0x00007ffff5ea4acd in  () at /usr/lib/libQt5Core.so.5
#17 0x00007ffff38b108c in start_thread () at /usr/lib/libpthread.so.0
#18 0x00007ffff5595e7f in clone () at /usr/lib/libc.so.6
(gdb) list
231	}
232	
233	// Filter with fast convolution (overlap-add algorithm).
234	int fftfilt::runFilt(const cmplx & in, cmplx **out)
235	{
236		data[inptr++] = in;
237		if (inptr < flen2)
238			return 0;
239		inptr = 0;
240	
(gdb) p data
$1 = (fftfilt::cmplx *) 0x285000002bc
(gdb) p inptr
$2 = 1
(gdb) p in
$3 = (const fftfilt::cmplx &) @0x7fff99490720: {_M_value = 0.00833129883 + -0.117736816 * I}
(gdb) p data[1]
Cannot access memory at address 0x285000002c4
(gdb) p data[0]
Cannot access memory at address 0x285000002bc
~~~

f4exb, 2018-04-18:
This affects in fact any plugin that uses the fftfilt class. There are cases where the filter is deleted early in the destructor before the threaded sink is deleted and therefore the feed method can still be called and the filter used. Sometimes the filter is not deleted in the destructor and therefore a memory leak will occur.

Potentially affected plugins are:
  - ChannelAnalyzer
  - ChannelAnalyzerNG
  - ATVDemod
  - BFMDemod (as reported here)
  - DATVDemod
  - SSBDemod
  - WFMDemod
  - TCPSrc
  - UDPSrc
  - ATVMod
  - SSBMod
  - WFMMod
  - UDPSink

f4exb, 2018-04-20:
Fixed in release v3.14.3

closed 2018-04-20 by f4exb