f4exb/sdrangel #160: Buffered squelch

f4exb, 2018-04-20:
The idea comes from here: https://brmlab.cz/user/jenda/kukuruku (See "Squelch sucks") 

It is actually an interesting idea to use a ring buffer on the audio to enhance the time when the squelch opens or closes according to the transmission at the expense of a delay proportional to the time of observation (gate time) of the squelch.

When the decision is made to open or close the squelch this is usually too late because you have taken the gate delay amount of time to decide to open or close but the actual transmission has started or stopped at the start of this delay period. A ring buffer can be used to recover the lost samples on the opening side and on the other end close immediately when the close decision is taken throwing away the dubious samples.

texasyojimbo, 2018-04-20:
It seems to me that buffering the audio could also allow for some later extensions/modifications/plugins that might allow for new and potentially-useful kinds of squelch control.

For example, open the squelch only if a certain audio sequence is detected (DTMF sequence, AX.25 packet header, etc.). Or for that matter, block digital signals only but pass analog ones (for example for Yaesu System Fusion or channels that are shared by both analog and digital systems; my county's fire/EMS repeater actually has both analog and digital systems on it).

Not asking that any of *that* actually be implemented, just saying that buffering a little audio could be a good architectural decision that might open some doors up later.

f4exb, 2018-04-21:
If you would like to open on a DTMF sequence or AX25 packet header then you might not want to hear that sequence so buffering is not useful in this case. 

However this is interesting for sorting out digital and analog signals because then you need some time to recognize the digital header and decide that this is a digital transmission. If it was an analog signal you would have lost the amount of samples needed to process the digital header. 

Practically this cannot be done in the NFM demod because it does not have all the DV logic so this could be an extension of the DSD demod that would pass through as analog FM anything it does not recognize as a DV signal. This goes beyond just squelch processing and could be the theme of another issue. I also find the analog/digital mixed repeaters annoying if you don't have the logic to sort out digital and analog signals.