f4exb/sdrangel #161: PTT button

bdn76, 2018-04-20:
Extremely nice soft. Cool! But what about PTT button for auto switch R/T for single device (HackRF for example)? Switch between tab, play stop, play start and play stop again very annoing.

f4exb, 2018-04-22:
You will have to use the REST API for this and write a script that executes the R/T switchover commands. Then this script can be triggered within the application using the commands feature. You can see an example of such a script written in Python here: https://github.com/f4exb/sdrangel/blob/master/swagger/sdrangel/examples/ptt.py 

This question has already been raised and discussed extensively. To summarize this is addressed with REST API and command feature which is described here: https://github.com/f4exb/sdrangel/tree/master/sdrgui#5b-commands

The REST API is not complete but BladeRF, HackRF, LimeSDR and RTL-SDR interfaces are already implemented.

bdn76, 2018-04-22:
ok

closed 2018-04-22 by bdn76