f4exb/sdrangel #165: HackRF decimation issue

bdn76, 2018-04-24:
I'm need monitor 2MHz spectrum and listen around 8 channel but i'm hear audio distortion with sample rate more than ~600ks/s but i need more for listen spectrum distnce channel (with SQ of course). Why i can not use high sample rate for NBFM without decimation?

f4exb, 2018-04-24:
I cannot understand this dribble can someone translate please?

f4exb, 2018-04-24:
Try to be more specific and describe exactly what you are trying to do with detailed steps to be able to reproduce.

bdn76, 2018-04-25:
I'm sorry for my english. In first example i use low decimation but nfm demodulator not work:

[image]

In second example i use high decimation and nfm demod work fine:

[image]

How i can listent to channel with ~1MHz distance?

f4exb, 2018-04-25:
Thanks for the screenshots this helps a lot. Well it looks like you are running a RTL-SDR not a HackRF. So I see a first issue with the settings of RTL-SDR: your gain is zero so you are not going to get much RF into the system anyway. The stripes you see on the waterfall are probably internal birdies. You should push this slider at least into the ~40 dB range and if you get overloaded by strong signals you can check the AGC checkbox next.

bdn76, 2018-04-25:
I'm provide with screenshot for explanation only because i have not HackRF at hand now but problem exactly the same - nbfm rx channel work fine with low I/Q sample rate (up to ~300k) and not work with high I/Q sample rate (above ~300k). I can listen one channel with low sample rate perfectly but i can not listen two channel simultaneously with big frequency distance (1MHz as on second screenshot). Are you understand my explanation?

f4exb, 2018-04-25:
Does it make any difference if at "high" I/Q sample rate (say 2.4 MS/s) you reduce the distance between channels to the same distance you are using for ~300k (with decimation)? This is to sort out if pb is due to sample rate or channel separation. Also can you provide a screenshot with the settings of the NFM demodulator?

bdn76, 2018-04-25:
I'm check now - no. With "high" I/Q sample rate i can not start even one channel but i can start two or more with "low" I/Q sample rate. Are you need screentshot?

f4exb, 2018-04-25:
A few more questions first:
- On which hardware are you running
- I see you run on a Debian buster (testing) distribution. Did you compile from source or did you install the Debian release package?

bdn76, 2018-04-26:
i5 M520 2.4G, 8GB.
Compile from source.
With "high" I/Q sample rate i can not use nfm demod at all - noise and rattle instead of voice but with "low" I/Q nbfm work fine.

f4exb, 2018-04-26:
No problem here... (this is dev version but there are no differences with 3.14.4 for that matter).
[image]

bdn76, 2018-04-27:
I'm think this is phosphor issue. If i'm enable phosphor then CPU usage drop to 200% insead of 400% without phosphor.

Work fine (~400% CPU usage, all 4 core work):
[image]
[image]

Not work (~200% CPU usage, only 2 core work):
[image]
[image]

And thus if i use only 2 core i have I/Q processing limitation.

f4exb, 2018-04-27:
OK I see... you have to set parameters sensibly. It does not make sense to run at 20 MS/s and have a RF filter at 1.75 MHz. 20 MS/s is very fast and needs powerful hardware (at least i7 8 core CPU) to be run properly. It is interesting to notice however that the phosphor thing changes the way the CPU cores are used. For sure 209.6% for 2 CPU is too much while 379% for 4 CPU is still (barely) acceptable.

This is very high CPU usage though. Did you compile with the 24 bit sample option on? You should turn it off for such high speed operation and of course compile for release and not debug.

f4exb, 2018-05-11:
Not an issue.

closed 2018-05-11 by f4exb