f4exb/sdrangel #168: typing a value to Δf in NFN demodulator takes wrong values

beta-tester, 2018-05-07:
in NFN Demodulator, when i type in a value to the Δf field i get wrong numbers, when the negative sign is selected.

e.g.:
1. case with no issue - sign is positive (+):
2. i click to the very left Digit and type in 1111111
3. in the Delta f it Shows 1,111,111 Hz

thats ok.

1. case with the issue . sign is negative (-):
2. i click to the sign in front of the very left Digit and Change it from positive to negative.
3. i click to the very left Digit and type in 1111111
4. in the Delta f it Shows me 8,440,505 Hz

tha's totally wrong.

[image]

texasyojimbo, 2018-05-07:
Hi beta-tester,

This is similar to #97 which I filed some time ago.

As I recall, we aren't really intended to do direct frequency entry here; this was originally conceived as something that would be adjusted only using up/down arrows or mouse. It is sort of a happy accident that it even somewhat works at all.

texasyojimbo, 2018-05-07:
Also, which version are you using, what platform, etc? Keep in mind that if Eduoard accepts this as a bug, he will probably need to reproduce it first. =D

beta-tester, 2018-05-07:
sorry, that i didn't gave any information about my environment i used.
OS: Microsoft Windows 10 Pro 64bit (1803)
SDRAngel: [sdrangel_v3.14.5.7z](https://github.com/f4exb/sdrangel/releases/download/v3.14.5/sdrangel_v3.14.5.7z)
SDR: RTL-SDR dongle or PlutoSDR

i am using the "direct input" by typing in the frequencies relative often, when i have a setup, where i have multiple decoders active to be able to follow different frequencies at the same time of the current bandwidth.
e.g.
NFN Decoder A: with Δf -0,012,500 Hz
NFN Decoder B: with Δf 0,000,000 Hz
NFN Decoder C: with Δf 0,012,500 Hz

f4exb, 2018-05-08:
Hi,
indeed there seems to be an issue with typing in negative values in the ValueDialZ component.

beta-tester, 2018-05-08:
thank you.
i could compile branch DEV on ubuntu 18.04 LTS and there it works.

beta-tester, 2018-05-08:
a little tiny issue still there. if the current value is +0,000,000 Hz then you chant change the sign, it stays at (+).
you have first to ave at least a number different to 0,000,000 to be able to type in a minus sign (-).
i think it comes, because when you out in a -0,000,000 it will change automatically to +0,000,000 (because by definition there is no -0, only +0)

f4exb, 2018-05-08:
Too complicated... you will have to live with this.