f4exb/sdrangel #169: DAB/DAB+ plugin would be nice

beta-tester, 2018-05-07:
i have a feature request.

it would be nice to have a DAB/DAB+ plugin available for sdrangle.
see [welle.io](https://github.com/AlbrechtL/welle.io)

f4exb, 2018-05-07:
You are welcome to contribute.

texasyojimbo, 2018-05-10:
This kind of dovetails with my desire to have a "how to" guide for building/writing new modules.

I never got around to setting up on Linux, but I will try to get that done at some point. You had suggested in an old issue that we start with the SSB module as a template.

f4exb, 2018-05-11:
I will have a look at welle.io though as suggested. I always have found more difficult to adapt existing software than starting from scratch. The interesting point with welle.io is that it is Qt based.

mpbraendli, 2018-09-02:
welle-io diverged a bit from SDR-J: I've split GUI and backend a bit better, there's a way to instantiate the backend without a GUI and without Qt, and it is able to decode audio from all the programmes in the multiplex at the same time.

All this lives in the `next` branch for now. Let me know if you use it!

texasyojimbo, 2019-05-09:
In addition to welle, there is also this, referenced in context of DAB transmission on the rtl-sdr.com blog today:

https://www.rtl-sdr.com/creating-a-dab-radio-station-with-a-limesdr/
https://github.com/Opendigitalradio

The rtl-sdr.com blog article mentions using welle for receiving, but the ODR people also have a Linux tool using GTK+ called dablin for reception.

f4exb, 2021-04-19:
Deployed in v6.10.0

closed 2021-04-19 by f4exb