f4exb/sdrangel #170: add timestamp to filename of recording sdriq file

beta-tester, 2018-05-07:
it would be great, when each recording to file would get a unique file name each time you press record.
it happend to me very often that i override an existing recorded sdrif file by accident, because each time sdrangel is using the same file name.

maybe  there is a reason for that current behavior, but there is also a reason to do not override files or better create unique files each time you make a record.
so maybe make it optionional to add a timestamp to the filename.

beta-tester, 2018-05-10:
thank you for the changed file name.
but guess the ':' (collon) in the filename (HH:mm:ss) will be an issue for windows file systems. a collon is not allowed - only for drive letter separation (**C:**\...)

f4exb, 2018-05-11:
This is why I like Windows so much :) The format `yyyy-MM-ddTHH:mm:ss` is the ISO format for a date/time stamp. OK to remove the colons ...

beta-tester, 2018-05-11:
oops...
i thought i could help a little bit with doing some easy stuff instead of wasting your time...
by doing [pullrequest ](https://github.com/f4exb/sdrangel/pull/177)...
but i was too slow.

closed 2018-05-11 by beta-tester

f4exb, 2018-05-11:
kept!