f4exb/sdrangel #172: PlutoSDR issue with soft decimation and Fc

beta-tester, 2018-05-08:
hello,
i have an issue with PlutoSDR and "Software Decimation" in combination with "Relative position of device center frequency".
it is not an issue with RTLSDR source, when using the equivalent switches.

if **Software Decimation** is =1, it does not matter that **Relative position of device center frequency** mode i use, all frequencies are where they should be.

if i turn **Software Decimation** >1 and **Relative position of device center frequency** mode to "Cen" then still everything is ok.

but if i turn **Software Decimation** to >1 and **Relative position of device center frequency** mode to "Sup" or "Inf", then the all frequencies are shifted or swapped  (don't know how to explain).

see picture compilation:
[image]
[image]
[image]
[image]
[image]
[image]

and in case **Software Decimation** =1, and **Relative position of device center frequency** mode "Inf" or "Sup", when you just switch from **Software Decimation** =1 to **Software Decimation** >1, then there it appears as when the **Relative position of device center frequency** is switched to "Cen" in reallity, because there is not that issue describes for "Inf"/"Sup", but it does not show "Cen" - until you touch the **Relative position of device center frequency** mode switch and switch it to something else and back to the "Inf" or Sup" mode, then you again will see issue in the frequency band.

f4exb, 2018-05-09:
By comparing `plutosdrinput.cpp` and `rtlsdrinput.cpp` it seems the logic implementing the frequency shift is not implemented for Pluto.

I would like to take this opportunity to challenge the way the shifted decimation is made. There is a possible optimization by using the rotating halfband decimators only or mostly. The purpose of this kind of operation was to move the DC out of the band. It is enough in fact to move it on one side in which case a successive combination of rotating halfband decimators could work. This makes the frequency shift calculation different but only determined by the decimation factor and the decimation chain implemented for this decimation factor.

f4exb, 2018-05-09:
These decimators have poor rejection and normally they should be used for monitoring a transmitter rather than for reception.

closed 2018-05-11 by beta-tester

f4exb, 2018-05-11:
This was not mentioned but v3.14.6 fixes the filtering also so inf/sup decimation can now be used normally. Moreover the frequency shift scheme has been changed to be closer to the device center frequency thus avoiding excessively large RF filters.

beta-tester, 2018-05-11:
i saw it... thanks.