f4exb/sdrangel #179: missing and blurry sdrangel application icon

beta-tester, 2018-05-11:
hi,
sdrangle has no application icon in the file browser on **Windows**.
[image]

and a blurry icon on **Ubuntu** and **Windows**, in the "_Task_" bar when the sdrangel application is running.
[image]

i guess it is because of the **sdrangel_icon.ico** is not a multi-resolution icon (i only 32x32) that makes it blurry when it is scaled up.

i have no experience with linux application icons at all.
i have no experience with windows application icons in Qt...
i have no idea where to put the icons to and how to link them to the project to get compiled/linked to the final executable application in a way, that is is used as application icon in the task bar at runtime and is visible in the file browser on Windows.
i only have experience with Windows application icons in VisualStudio, but that experience doesn't help me here.

so all my tries to provide useful code changes for a pull-request failed.
but i provide a zip file with files for a "_hires_" icon (at least for Windows).

- **sdrangel_icon_template (needs font OpenSans Bold).svg** is the base template for all folowing icon files. it requires "**Open Sans** (Bold)" font you can download [here](https://fonts.google.com/specimen/Open+Sans)
- **sdrangel_icon_scalable.svg** is for linux (ubuntu **/usr/share/icons/hicolor/scalable/**). it does not need any font.
- **sdrangel_icon.xcf** is the template for multi-resolution **sdrangel_icon.ico**
- **sdrangel_icon.ico** is the multi-resolution icon for at least Windows application.
- **sdrangel_icon_256x256.png** ... **sdrangel_icon_16x16.png** are the icons in the most common resolution (ubuntu **/usr/share/icons/hicolor/256x256/** ... **/usr/share/icons/hicolor/16x16/**).

[sdrangel_icons.zip](https://github.com/f4exb/sdrangel/files/1996428/sdrangel_icons.zip)

Vort, 2019-05-22:
Strange thing is that project contains `sdrangel_icon.ico`, `sdrangel.rc` and corresponding line in `CMakeLists.txt`.
In my custom build icon also appears. So problem with missing icon is somewhere in build process.

f4exb, 2019-05-22:
The build process is being revisited thanks to the contribution of ra1nb0w. Currently build on windows is hand crafted without cmake.

Vort, 2019-05-22:
You mean `.pro` files?
[Adding of](https://doc.qt.io/qt-5/appicon.html) `RC_ICONS = ../sdrgui/resources/sdrangel_icon.ico` line into `app.pro` should be enough.

f4exb, 2019-05-22:
Thanks! It's working

beta-tester, 2019-06-04:
thank you for fixing the icon.

closed 2019-06-04 by beta-tester