f4exb/sdrangel #181: datvideorender.cpp Error during make

cpbridges, 2018-05-25:
Hi F4EXB,

I can into the following error during a build on fresh Ubuntu 16.04 LTS. All dependancies already installed including for rtl and pluto. Any ideas?

```
[ 60%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir
...
/datvideorender.cpp.o
/home/chrisbridges/sdrangel/plugins/channelrx/demoddatv/datvideorender.cpp:199:105: error: ‘av_dict_get_string’ was not declared in this scope
   av_dict_get_string(m_objFormatCtx->programs[m_intVideoStreamIndex]->metadata,&objBuffer,':','\n');
                                                                                                   ^
/home/chrisbridges/sdrangel/plugins/channelrx/demoddatv/datvideorender.cpp:208:100: error: ‘av_dict_get_string’ was not declared in this scope
    av_dict_get_string(m_objFormatCtx->streams[m_intVideoStreamIndex]->metadata,&objBuffer,':','\n');
                                                                                                   ^
```
73s Chris

f4exb, 2018-05-25:
Did you install all dependencies? More specifically: `sudo apt-get install  ffmpeg libavcodec-dev libavformat-dev`

FFY00, 2018-05-26:
Hi, I am running into this same issue on my machine, which is running Arch Linux. I have `ffmpeg 4.0` installed. The `ffmpeg` package provides `libavcodec.so=58-64` and `libavformat.so=58-64`.

f4exb, 2018-05-27:
`av_dict_get_string` is a symbol of FFmpeg library. If your installation cannot provide the right FFMmpeg there is not a lot I can do about this. If you don't need the DATV demodulator then just comment out the relevant line in https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/CMakeLists.txt

cpbridges, 2018-05-27:
OK - I have it also installed. Will give it a recheck & report back. 73s C

cpbridges, 2018-05-27:
```
$ sudo apt-get install ffmpeg libavcodec-dev libavformat-dev 
[sudo] password for chrisbridges: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ffmpeg is already the newest version (7:2.8.14-0ubuntu0.16.04.1).
libavcodec-dev is already the newest version (7:2.8.14-0ubuntu0.16.04.1).
libavformat-dev is already the newest version (7:2.8.14-0ubuntu0.16.04.1).
```
As this is still unresolved, I'll keep at this error. Thoughts / recommendations welcome!

f4exb, 2018-05-28:
`sudo apt-get install ffmpeg libavcodec-dev libavformat-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ffmpeg is already the newest version (7:3.4.2-2).
libavcodec-dev is already the newest version (7:3.4.2-2).
libavformat-dev is already the newest version (7:3.4.2-2).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.`

This is on Ubuntu 18.04. No issues.

f4exb, 2018-05-28:
The AVUTIL cmake variables may be missing. Can you try to apply the following patch:

```
diff --git a/plugins/channelrx/demoddatv/CMakeLists.txt b/plugins/channelrx/demoddatv/CMakeLists.txt
index f12dac3f..bf41c23f 100644
--- a/plugins/channelrx/demoddatv/CMakeLists.txt
+++ b/plugins/channelrx/demoddatv/CMakeLists.txt
@@ -25,6 +25,7 @@ include_directories(
        ${CMAKE_CURRENT_BINARY_DIR}
        ${AVCODEC_INCLUDE_DIRS}
        ${AVFORMAT_INCLUDE_DIRS}
+       ${AVUTIL_INCLUDE_DIRS}
        ${SWSCALE_INCLUDE_DIRS}
 )
 
@@ -48,6 +49,7 @@ target_link_libraries(demoddatv
        sdrgui
        ${AVCODEC_LIBRARIES}
        ${AVFORMAT_LIBRARIES}
+       ${AVUTIL_LIBRARIES}
        ${SWSCALE_LIBRARIES}
 )

```

f4exb, 2018-05-29:
`-- Found FFmpeg: /usr/local/lib/libavcodec.a;/usr/local/lib/libavformat.a;/usr/local/lib/libavutil.a` 
These are not the places where Ubuntu installs FFmpeg moreover it uses shared libraries.

closed 2018-07-03 by f4exb