f4exb/sdrangel #182: DVB-T(2) demodulator

Yume303, 2018-05-27:
It would be possible to add DVB-T(2) demodulator ?

veso266, 2019-01-12:
well DVB-S is added, maybe DVB-T could be next then DVB-T2, then DVB-S2 and maybe maybe DVB-C

KC7NOA, 2019-01-15:
Wonderful news

zanco, 2019-02-21:
Hi, with the Es'Hail satellite activated and DVB-S2 choosen as standard, is there a change that DVB-S2 is added to the software as reception mode ?
Thanks,
Ben

KC7NOA, 2019-02-26:
I dont see DVB-S2 as standard ... listed as experimental .... from AMSAT-UK

Es’hail-2 carries two “Phase 4” amateur radio transponders<https://ukamsat.files.wordpress.com/2018/11/amsat-uk_eshail-2_transponder_info.pdf> operating in the 2400 MHz and 10450 MHz bands. A 250 kHz bandwidth linear transponder intended for conventional analogue operations and an 8 MHz bandwidth transponder for experimental digital modulation schemes and DVB amateur television.

KC7NOA

f4exb, 2022-02-11:
dropped

closed 2022-02-11 by f4exb